NanoVNA-Saver/nanovna-saver #489: NanoVNA-Saver version 0.4.0 Marker 2 Error

PA3CCE, 2022-04-12:
I`m measuring the response of a filter (S21 Gain), using NanoVNA-Saver V0.4.0 and my LiteVNA..
The markers 1 and 3 are shown in the graph , but marker 2 is not visible, the data is shown correctly.

gaionim, 2022-05-04:
Can you try to change color and/or increase dimension ?

PA3CCE, 2022-05-05:
Hi Gaionim,
I remamber the demensions of markes 1 and 3 wre VERY tiny at that moment.
I don`t have the filter I measured before at hand at the moment, so I just measured the S21 of a short (uncalibrated) and this time all three markers show up  and are NOT THAT tiny as previously, so ???
I will  do some more testing and will let you know when I can reproduce the error.
Sorry

gaionim, 2022-05-05:
I think the problem not is in the data, so i guess it is resolved ;-)

For me was the combination of small dimension of the marker AND the color;
changing color or increasing dimension solved the issue ( at least in my case ).

If no more problem, can you close the issue ?
You'll can reopen if/when returns.

PA3CCE, 2022-05-05:
Yes I will close the issue, when it shows again I will reopen it..
To me it looks like it was related to the very tiny demensions , they were extreem small indeed originally, hardly visible..
Strange , I id not chance any setting !!

closed 2022-05-06 by PA3CCE