NanoVNA-Saver/nanovna-saver #494: Enhanced Response / "thrureflr" Calibration Needs an Explaination in Documentation

biergaizi, 2022-05-01:
In NanoVNA Saver 0.4.0, Enhanced Response calibration has been [added](https://github.com/NanoVNA-Saver/nanovna-saver/commit/915da14ac1abc7af4da8dd92b3c7c3aa386c68cf), allowing for more accurate measurements. Unfortunately, I don't see any documentation that describes this calibration.

In particular, I already have existing experience doing the *Enhanced Response* calibration. Before this feature was added, I already did this calibration myself using some code I wrote for data postprocessing with the scikit-rf library. Yet, I have no idea about what the "thrureflr" calibration step means. I searched the web search in quote marks and I do not see a single mention of this word in anywhere on the Internet.

My suspicious is that it's a calibration for the imperfection of the "thru" standard. Thus, If I'm measuring an [Insertable Device](https://na.support.keysight.com/pna/help/latest/S3_Cals/Calibration_THRU_Methods.htm), I can assume the "thru" is zero-length (flush thru), so I can just reuse the same measurement for the "thru" step for the "thrureflr" step. Is it correct?

**Describe the solution you'd like**
The documentation, such as README, should be updated with a brief explanation of what the user should do in the "thrureflr" calibration step.

PA3CCE, 2022-05-01:
I found that the "thrureflr" is filled automatically after doing a normal "Through" calibration, so without doing anything extra myself.
I as well , would like to know what has happened to improve the calibration (if there has been inprovement indeed)  !
Thanks.

biergaizi, 2022-05-01:
@PA3CCE: I didn't notice the auto-fill behavior. Thanks for pointing it out.

BTW, If you just want to know how does Enhanced Response improves calibration, you can read the documentation of scikit-rf, [Calibration With Three Receivers](https://scikit-rf.readthedocs.io/en/latest/examples/metrology/Calibration%20With%20Three%20Receivers.html) and [TwoPortOnePath, EnhancedResponse, and FakeFlip](https://scikit-rf.readthedocs.io/en/latest/examples/metrology/TwoPortOnePath%2C%20EnhancedResponse%2C%20and%20FakeFlip.html). The math is also briefly explained in the NanoVNA mailing list, in the thread [Enhanced Response for S21 measurements](https://nanorfe.com/forum/Enhnanced-Response-for-S21-measurements.html)

Basically, in a complete, 2-port, 2-path VNA. When you perform a measurement, the calibration is applied to both the forward direction (S11 and S21) and the reverse direction (S22 and S12), thus the imperfections at both ports are completely removed using a 12-term error model. However, for a 2-port, 1-path VNA like the NanoVNA, it cannot measure the reverse direction and a full error correction is not possible.

In this situation you basically have three choices. First, just do a simple response calibration and normalize the insertion loss, this does not provide any vector error correction. Second, you can assume the port 2 is perfectly matched and ignore that term in the error model, then a partial error correction is possible again. If the mismatch at port 2 is small, the data quality can be pretty good. This calibration is called Enhanced Response. 

In the previous NanoVNA Saver code, the calibration implementation is a bit flawed. If I get it correct, it uses the same error term e11 for port 1 to port 2. This creates a significant ripple around 0.2 dB in the S21 data. The updated code now correctly set e22 to 0. Now it becomes a proper Enhanced Response calibration and the ripple disappears.

P.S: The Enhanced Response itself is also a degenerate form of a full Two Port One Path calibration - even on a 1-path VNA a full error correction is possible still if you measure the device in both directions. Once the S11 and S21 are measured, you physically flip the device to do another measurement to get the missing S12 and S22. Now you can do a perfect error correction that is basically equivalent to the full 12-term error model. And if the device-under-test is reciprocal and symmetric (S11 = S22, S12 = S21), e.g. many passive LCR circuits, you can enjoy full error correction by copying S11 to S22, S21 to S12, even without flipping it (scikit-rf calls this trick "FakeFlip"). Both techniques are correctly not implemented in NanoVNA Saver, I'm currently doing it myself programmatically during data postprocessing. It would be interesting if someone adds this as a special feature to NanoVNA Saver with a GUI wizard.

biergaizi, 2022-05-01:
Update: I made some edits to my original comments to correct mistakes. If you're subscribed to this issue via E-mail, please take a look at the web page (in the past, I've experienced some misunderstanding because of this).

PA3CCE, 2022-05-01:
Thank you so much Biergaizi fo alli nfo.

closed 2022-05-16 by zarath