NanoVNA-Saver/nanovna-saver #502: Errors during device enumeration on startup crashes the program

jeffpc, 2022-05-22:
**Describe the bug**
Errors during device enumeration on startup crashes the program on FreeBSD.

**To Reproduce**
Steps to reproduce the behavior:
1. plug in some serial USB device (becomes cuaU0)
2. plug in NanoVNA (becomes cuaU1)
3. run NanoVNA-Saver
4. see output on console failing to open cuaU0 preventing program start

```
NanoVNASaver 0.3.10

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020, 2021 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

2022-05-22 11:39:16,739 - NanoVNASaver.Hardware.Hardware - ERROR - No VNA detected. Hardware responded to CR with: 
2022-05-22 11:39:17,168 - NanoVNASaver.Hardware.Hardware - WARNING - Did not recognize NanoVNA type from firmware.
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
OSError: [Errno 16] Device busy: '/dev/cuaU0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/NanoVNASaver", line 33, in <module>
    sys.exit(load_entry_point('NanoVNASaver==0.3.10', 'console_scripts', 'NanoVNASaver')())
  File "/usr/local/lib/python3.8/site-packages/NanoVNASaver/__main__.py", line 85, in main
    window = NanoVNASaver()
  File "/usr/local/lib/python3.8/site-packages/NanoVNASaver/NanoVNASaver.py", line 92, in __init__
    self.serial_control = SerialControl(self)
  File "/usr/local/lib/python3.8/site-packages/NanoVNASaver/Controls/SerialControl.py", line 37, in __init__
    self.rescanSerialPort()
  File "/usr/local/lib/python3.8/site-packages/NanoVNASaver/Controls/SerialControl.py", line 67, in rescanSerialPort
    for iface in get_interfaces():
  File "/usr/local/lib/python3.8/site-packages/NanoVNASaver/Hardware/Hardware.py", line 96, in get_interfaces
    iface.open()
  File "/usr/local/lib/python3.8/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 16] could not open port /dev/cuaU0: [Errno 16] Device busy: '/dev/cuaU0'
```

If I plug in NanoVNA only (so it becomes cuaU0), NanoVNA-Saver starts up and functions fine.

**Desktop (please complete the following information):**
 - OS: FreeBSD 13.0-RELEASE-p11
 - Python version: 3.8
 - NanoVNA-Saver version: 0.3.10 (FreeBSD package)

zarath, 2022-05-25:
Hey @jeffpc, can you check-out #504

jeffpc, 2022-05-26:
Sorry for the delay. I'll try to give it a try tonight.

jeffpc, 2022-05-27:
It turns out that FreeBSD packages have two small patches and an extra file [1]. Part of those changes are related to getting the vid/pid of the device.  Without those changes, the vid/pid are None, and so they never match and the devices are ignored.  (It turns out that there is bug in that code and so I just filed a FreeBSD bug about it [2].)

Now I'm thinking that the serial exception I saw was because of the FreeBSD bug - nanovna-saver found cuaU0 (non-VNA device), but got the wrong vid/pid because of the bug, then tried to open the device and got EBUSY.

However, I think your fix is still needed because the VNA device could legitimately be open by another process (producing an EBUSY) or the device node could be not accessible by the current user (producing an EACCES).  If I make the device file inaccessible, I get (without your fix):

```
...same stack as before...
serial.serialutil.SerialException: [Errno 13] could not open port /dev/cuaU0: [Errno 13] Permission denied: '/dev/cuaU0'
```

With your fix, the exception goes away and NanoVNA-Saver starts up fine.

[1] https://cgit.freebsd.org/ports/tree/comms/nanovna-saver/files
[2] https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=264271

commit ee3048d9 (2022-05-27): catch serial exception on device open - fixes #502 (#504)
closed 2022-05-27 by zarath