NanoVNA-Saver/nanovna-saver #511: May 29, 2022 commit results python error.

w5pny, 2022-05-30:
Ubvuntu 20.04 python 3.8

With the May 29 2022 commit, I always get:

Traceback (most recent call last):
  File "/usr/local/lib/nanovnasaver/nanovna-saver.py", line 26, in <module>
    from NanoVNASaver.__main__ import main
  File "/usr/local/lib/nanovnasaver/NanoVNASaver/__main__.py", line 36, in <module>
    from NanoVNASaver.NanoVNASaver import NanoVNASaver
  File "/usr/local/lib/nanovnasaver/NanoVNASaver/NanoVNASaver.py", line 28, in <module>
    from .Windows import (
  File "/usr/local/lib/nanovnasaver/NanoVNASaver/Windows/__init__.py", line 6, in <module>
    from .DisplaySettings import DisplaySettingsWindow
  File "/usr/local/lib/nanovnasaver/NanoVNASaver/Windows/DisplaySettings.py", line 25, in <module>
    from NanoVNASaver.Charts.Chart import (
  File "/usr/local/lib/nanovnasaver/NanoVNASaver/Charts/__init__.py", line 2, in <module>
    from .Frequency import FrequencyChart
  File "/usr/local/lib/nanovnasaver/NanoVNASaver/Charts/Frequency.py", line 37, in <module>
    class FrequencyChart(Chart):
  File "/usr/local/lib/nanovnasaver/NanoVNASaver/Charts/Frequency.py", line 453, in FrequencyChart
    def _data_oob(self, data: list[Datapoint]) -> bool:
TypeError: 'type' object is not subscriptable

I don't know if this is sufficient, but it seems to clear the error so far:

--- ./NanoVNASaver/Charts/Frequency.py.orig	2022-05-29 10:40:50.000000000 -0600
+++ ./NanoVNASaver/Charts/Frequency.py	2022-05-30 10:39:48.863412359 -0600
@@ -450,7 +450,7 @@
             self.drawDragbog(qp)
         qp.end()
 
-    def _data_oob(self, data: list[Datapoint]) -> bool:
+    def _data_oob(self, data: list) -> bool:
         return (data[0].freq > self.fstop or self.data[-1].freq < self.fstart)
 
     def _check_frequency_boundaries(self, qp: QtGui.QPainter):

zarath, 2022-09-20:
Could be related by leaked numpy arrays to locations where python array is expected.
Stil valid with v0.5.3?

mercy2ro, 2022-09-28:
I think this is because the difference between python3.8 and python3.9.

https://peps.python.org/pep-0585/
https://github.com/NanoVNA-Saver/nanovna-saver/issues/522

So this issue will be solved by the following change.

def _data_oob(self, data: list[Datapoint]) -> bool:
↓
def _data_oob(self, data: List[Datapoint]) -> bool:

After this change, the error went away. (In my python3.8 environment)

closed 2022-09-29 by zarath