NanoVNA-Saver/nanovna-saver #517: Linux: "standard" way (PPA, .deb) to install NanoVNA-Saver?

Ho-Ro, 2022-07-12:
**See [this request](https://groups.io/g/nanovna-users/topic/92215641#28696) from NanoVNA user group:**
*"Hello, is there possible to use a "standard" way (PPA, .deb) to install NanoVNA-Saver?
Thanks"*

**Describe the solution you'd like**
I build a deb package with this simple command:
```
DEB_BUILD_OPTIONS=nocheck python3 setup.py --command-packages=stdeb.command bdist_deb
```
This command could go into a `Makefile` to ease the build and install process.

You need the debian package `python3-stdeb` (Python to Debian source package conversion plugins for distutils) and maybe some more development packages. 
For a similar project with github actions I do this build setup:
```
sudo apt-get install --quiet --yes fakeroot python3-setuptools python3-stdeb dh-python
```
As a quick start see [this documentation](https://pypi.org/project/stdeb/).

Ho-Ro, 2022-10-11:
Ok, @zarath would you like me to provide a PR? It's a simple enhancement for Debian (and also Ubuntu) users without destroying the Win etc. functionality.

zarath, 2022-10-11:
with pleasure

Ho-Ro, 2023-01-10:
Ok, implemented with #560 and #662 - ca be closed.

closed 2023-01-10 by Ho-Ro