NanoVNA-Saver/nanovna-saver #522: NanoVNA-Saver crashes on startup

radi8, 2022-07-22:
Since the Hotfix commit I get this error on starting nanoVNA-Saver. If I checkout the 'Hotfix' sha, everything runs just fine.

/opt/nanovna-saver$ python3 nanovna-saver.py
Traceback (most recent call last):
  File "nanovna-saver.py", line 26, in <module>
    from NanoVNASaver.__main__ import main
  File "/opt/nanovna-saver/NanoVNASaver/__main__.py", line 36, in <module>
    from NanoVNASaver.NanoVNASaver import NanoVNASaver
  File "/opt/nanovna-saver/NanoVNASaver/NanoVNASaver.py", line 28, in <module>
    from .Windows import (
  File "/opt/nanovna-saver/NanoVNASaver/Windows/__init__.py", line 6, in <module>
    from .DisplaySettings import DisplaySettingsWindow
  File "/opt/nanovna-saver/NanoVNASaver/Windows/DisplaySettings.py", line 25, in <module>
    from NanoVNASaver.Charts.Chart import (
  File "/opt/nanovna-saver/NanoVNASaver/Charts/__init__.py", line 2, in <module>
    from .Frequency import FrequencyChart
  File "/opt/nanovna-saver/NanoVNASaver/Charts/Frequency.py", line 37, in <module>
    class FrequencyChart(Chart):
  File "/opt/nanovna-saver/NanoVNASaver/Charts/Frequency.py", line 453, in FrequencyChart
    def _data_oob(self, data: list[Datapoint]) -> bool:
TypeError: 'type' object is not subscriptable
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 153, in apport_excepthook
    with os.fdopen(os.open(pr_filename,
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_opt_nanovna-saver_nanovna-saver.py.1000.crash'

Original exception was:
Traceback (most recent call last):
  File "nanovna-saver.py", line 26, in <module>
    from NanoVNASaver.__main__ import main
  File "/opt/nanovna-saver/NanoVNASaver/__main__.py", line 36, in <module>
    from NanoVNASaver.NanoVNASaver import NanoVNASaver
  File "/opt/nanovna-saver/NanoVNASaver/NanoVNASaver.py", line 28, in <module>
    from .Windows import (
  File "/opt/nanovna-saver/NanoVNASaver/Windows/__init__.py", line 6, in <module>
    from .DisplaySettings import DisplaySettingsWindow
  File "/opt/nanovna-saver/NanoVNASaver/Windows/DisplaySettings.py", line 25, in <module>
    from NanoVNASaver.Charts.Chart import (
  File "/opt/nanovna-saver/NanoVNASaver/Charts/__init__.py", line 2, in <module>
    from .Frequency import FrequencyChart
  File "/opt/nanovna-saver/NanoVNASaver/Charts/Frequency.py", line 37, in <module>
    class FrequencyChart(Chart):
  File "/opt/nanovna-saver/NanoVNASaver/Charts/Frequency.py", line 453, in FrequencyChart
    def _data_oob(self, data: list[Datapoint]) -> bool:
TypeError: 'type' object is not subscriptable

radi8, 2022-07-22:
Sorry LinuxMint 20.3 on several different machines.

fredericgermain, 2022-08-10:
It seems that master won't work with python 3.8, since commit 2821f4d0 2022/05/28 at least.

python 3.8 is python3 on Ubuntu 20,04, which I'm using.

Solved by using python3.9 and a few steps. Readme on Ubuntu/linux should be updated most probably.

1. Install python3.9 and pip
```
sudo apt install python3.9 python3.9-venv -y
python3 -m venv ~/.venv_nano
 . ~/.venv_nano/bin/activate
pip install wheel # can probably be added in setup.cfg ? or that's install by the  pip install -U pip
```

3. Update pip and run the pip installation

```
python3.9 -m pip install .
```

4. Once completed run with the following command

```
 . ~/.venv_nano/bin/activate
python3.9 nanovna-saver.py
```

PR #554 (2022-09-29): bugfix for python3.8 compatability
fixes #522, #511
commit 35686319 (2022-09-29): bugfix for python3.8 compatability

fixes #522, #511
closed 2022-09-29 by zarath