NanoVNA-Saver/nanovna-saver #523: NanoVNASaver 0.4.0 fails calibration

heribertmsb, 2022-07-24:
**Describe the bug**
Calibartion ends after apply with 
[image]
[image]

 NanoVNA-Saver version: 

[image]

I have tried several bands, some are working, some not.
Also entering small deviation in frequency for the bandwith makes no change

Howeve calibration on the device works fine and thesweep can be loaded offline

kawalkQCWski, 2022-07-25:
can you try the latest firmware which is 1.04 ? 

https://deepelec.com/2022/04/02/deepvna_101_nanovna-f-1_0_4/

heribertmsb, 2022-07-26:
I have diffrent Hardware: NanoVNA-F V2 Hardware rev. C (no thumbweel switch) by sysjoint. 
Also no "by BH5HNU mark on the PCB 
Here the link to my hardware:  https://sysjoint.com/en/content/?144.html
lateset firmware: http://www.sysjoint.com/file/NanoVNA-F_V2_App_v0.5.0.zip

kawalkQCWski, 2022-07-26:
Thanks for that. I didn't realize that the V2 is a sysjoint product until yesterday,  Confusing naming.  Some people load the deepelec firmware into these but that creates other issues for the saver.  Not sure what if any hardware differences might be causing that as one seems to  be the clone of the other.  Maybe the processor used is not STM but alternative ?

heribertmsb, 2022-07-26:
In the manual i found the following about the hardware
[image]

kawalkQCWski, 2022-07-26:
Definitely not complete - as it's missing the Ato D converter that makes
the F  perform quite well.

[image: image.png]

kawalkQCWski, 2022-07-27:
VNA F 

[image]

VNA F-V2 

[image]

kawalkQCWski, 2022-07-27:
GD32 is almost the same but a little faster and has different flash implementation

kawalkQCWski, 2022-07-27:
Check this out  about the F-V2 and F 

https://github.com/NanoVNA-Saver/nanovna-saver/issues/523

closed 2022-11-26 by github-actions[bot]