NanoVNA-Saver/nanovna-saver #524: Unable to install on MacBook Air with M1 SoC

dwadi, 2022-08-01:
I have question how can i install this app on laptop with M1 SoC?

Ho-Ro, 2022-08-02:
You could install an up-to-date Python system with the [required modules](https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/requirements.txt) (dunno if Cython is needed), clone the project from GitHub and run [`nanovna-saver.py`](https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/nanovna-saver.py) directly from the project directory - this is how I do it on Linux - no need to install or build this app. The program may complain if some other modules are missing - just install them via pip or brew and retry.

zarath, 2022-09-18:
Since with v0.5.3 release it also should work with python 3.10

closed 2022-09-18 by zarath