NanoVNA-Saver/nanovna-saver #534: LiteVNA

PA3CCE, 2022-09-03:
Please add support for LiteVNA

zarath, 2022-09-07:
whatever that device will be...

DiSlord, 2022-09-11:
LiteVNA correct work on NanoVNA-Saver
LiteVNA use V2 protocol

PA3CCE, 2022-09-11:
@ Zarath
LiteVNA is a well known upgrade to NanoVNA.H/H4.
@ DiSlord
Thank you very much for the info, so the update will be very easy I suppose..

PA3CCE, 2022-09-29:
LiteVNA is recognized as NanoVNA V2, so just the name should be corrected.

zarath, 2023-03-02:
@PA3CCE do you have a debug output of the detection.
Is it possible to distinguish it from V2 on protocoll level?

PA3CCE, 2023-03-02:
@DiSlord
May I redirect this question to you,  being the author of the specific firmware ?
@Zarath
The LiteVNA has a much wider frequency range then the V2 but it looks like this is no prblem for the -Saver.
Thank you both

PA3CCE, 2023-03-04:
@Zarath:: Another diference in LiteVNA to NanoVNA V2 is that the number of measurements in LiteVNA can be set in units,  from 1 to 1024 included.
In that respect, see my errormessage #405 still pending !

PA3CCE, 2023-06-03:
Sorry, in the latest version the LiteVNA and it's settings still are not implemented !

Entropy512, 2023-08-02:
@zarath - I have a LiteVNA and might be able to work on this, however I have no familiarity with your specific codebase and what mechanisms you might already have for autodetection.

What do NanoVNA-V2 units report for the deviceVariant register?  http://nanovna.com/wp-content/uploads/2021/11/LiteVNA_User-Guide.pdf page 25

zarath, 2023-08-02:
@Entropy512 it reports the same values in deviceVariant register as an original S-A-A-2. So this class of devices is indistinguishable by client software. The document even tell, that this was intended. :-/

So the problem is just the wrong type name in the device selector? Or isn't it detected because it differs in USB_ID?

Entropy512, 2023-08-08:
It works, however it supports additional functionality not supported by the Nano (more than 101 sweep points per segment, although leveraging this would require some additional UI work beyond detecting the device as being different)

So it's kind of annoying that they intentionally made deviceVariant match another device - really????  :(

PA3CCE, 2023-08-09:
Yes indeed, the device connects to NanoVNA-Saver, but showing it being a V2 is not the biggest problem.
It works **more or less** using the V2 characteristics, but it has more and better functionalities.

Entropy512, 2023-08-09:
Among other things (which I am wondering if it might be a contributor to some of the issues I'm seeing):

It looks like https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/src/NanoVNASaver/Hardware/NanoVNA_V2.py will allow setting TX power when the frequency is below 3 GHz.  However, that register (0x42) is used on the LiteVNA to set the TX power for the high-frequency oscillator.  0x41 is used for the low-freq oscillator.  I don't recall at the moment where the unit switches between these two.  I need to do some debug runs to see if this is actually coming into play with my issues in the 5 GHz wifi band.

zarath, 2023-08-09:
perhaps one with a liteVNA can checkout https://github.com/zarath/nanovna-saver/tree/feature/clock_detect 
and do a debug output with nanovna-saver (-D debug.txt) and can attach the output on this discussion.
If the unix timer registers are readable it would be possible to distinguish the devices from the original ones.

PA3CCE, 2023-08-09:
Better ask Dislord indeed, he wrote the firmwares and knows all the ins and outs of the LiteVNA and the V2, he was so kind to make some adaptions into NanoVNA-APP!

PA3CCE, 2023-08-09:
@zarath
I would like to do so, but recently I "upgraded" from Windows10 to Windows11 and since then I have big problems with the serial communiication to several devices, thank you Microsoft !!
I have been working on it quite a while, but still no luck, sorry.

DiSlord, 2023-12-02:
Here is how NanoVNA-App detect V2 device:
01.323 tx: requesting device variant
01.323 tx: 10 F0
01.324 rx: 02 .. DEVICE VARIANT OK
01.324 tx: requesting serial protocol version
01.325 tx: 10 F1
01.384 rx: 01 .. PROTOCOL VERSION OK
01.385 tx: requesting hw-revision and fw-version
01.385 tx: 10 F2 10 F3 10 F4 10 40
01.386 rx: 02 ..... HARDWARE REVISION
01.387 rx: 02 02 .. FIRMWARE VERSION

Register F0 contain device revision, always 0x02 for V2 series
Refister F1 contain protocol version, always contain 0x01

Next registers depend from device:
F2 - BOARD_REVISION
0x02 for V2, LiteVNA
0x03 for V2Plus
0x04 for V2Plus4

F3 - FW_MAJOR_VER
0x01 for V2, V2Plus, V2Plus4
0x02 for LiteVNA

F4 - FM_MINOR_VER
0x02 - for all V2, V2Plus, V2Plus4, LiteVNA

LiteVNA and V2Plus4 allow 65535 points over USB
V2, V2Plus only 1024

DiSlord, 2023-12-03:
V2 Binary protocol:
V2 contain 256 registers for read/write

On write to registers data put to reg[address] and V2 grab data and process (depend from address)

All values after in hex format!!!!
Commands for read/write
0x0D - read version (return "2")
0x10 - read byte
0x11 - read word
0x12 - read dword
0x13 - read qword

0x18 - read measured data (use 30 address for count)

0x20 - write byte
0x21 - write word
0x22 - write dword
0x23 - write qword
0x28 - write buffer (next byte data size, and data)

address:
0x00 - (qword) start frequency
0x10 - (qword) frequency step
0x20 - (word) points count
0x22 - (word) values pre frequency (need set to 1, 

0x26 - (byte) mode (0 - USB mode, 1 - send raw data, 2 - normal mode, 3 - send calibrated data)
0x30 - (byte) points for measure, or up to (measured values)

0x40 - (byte) average
0x41 - (byte) si power
0x42 - (byte) adf power
0x44 - (byte) write channel select (1 - send only CH0 data, 2 send only CH1, any other - send both)

0x50 - (dword) color value
0x54 - (byte) color index (use H/H4 indexes)
0x58 - (dword) write set time in UNIX format
0x5C - (word) read battery value in millivolts

0xEE - (byte) any value, on write V2 send screenshot

0xD0 - (dword) SN[0] (serial number 0)
0xD4 - (dword) SN[1] (serial number 1)
0xD8 - (dword) SN[2] (serial number 2)

0xF0 - (byte) device variant (2 for V2)
0xF1 - (byte) protocol version (1 for V2)
0xF2 - (byte) board revision (2 for V2, 3 - V2Plus, 4 - V2Plus4)
0xF3 - (byte) firmware major (1 for all)
0xF4 - (byte) firmware minor

Example:
Set average = 5 command:
20 - write byte
40 - address (average)
05 - value
Need send : 204005

Set color[7]=RRGGBB:
20 - write byte
54 - address (color index)
07 - value (7 index is LCD_TRACE_2_COLOR)
22 - write dword
50 - address (color value)
BB - color in RGB
GG - color in RGB
RR - color in RGB
00 - 
Need send: 2054072250BBGGRR00

Read color from color[8]:
20 - write byte
54 - address (color index)
08 - value (8 index is LCD_TRACE_3_COLOR)
12 - read dword
50 - address (color value)
Need send: 2054081250
rx *00 fc 00 00
    BB GG RR 00

PA3CCE, 2023-12-07:
My USB problem on windows 11 solved : 
I found that Windows 11 is using the COM ports 4 and 7 for Bluetooth on my system (**Windows 10 did** **not**), this gives conflicts with previously installed USB devices.  So I have to **disconnect Bluetooth in** Windows 11 settings, to be able to use these devices again. **Very anoying !**
**@ Zarath** 
DiSlord has sent a lot of information (see above, **thank you so much DiSlord**), I hope, now you have all the information needed. to implement LiteVNA into NanoVNASaver.
Jos

PA3CCE, 2024-01-10:
I sincerely hope you will find it, it would be highly appreciated.
Please note : This is no complaint !!

akohlsmith, 2024-08-29:
I have been using nanovna-saver with my LiteVNA64 without issue. Results over USB appear to be calibrated, but I may be wrong. Is there a foolproof way to verify this?

PA3CCE, 2024-09-09:
> I have been using nanovna-saver with my LiteVNA64 without issue. Results over USB appear to be calibrated, but I may be wrong. Is there a foolproof way to verify this?
====================================

As said before, LiteVNA64 connects to NanoVNA-Saver as being a NanoVNA V2 and measures if is a V2, but 
 the LiteVNA64 performs better an HAS MORE FEATURES then the V2

PA3CCE, 2024-11-15:
> I'll have to find some time to implement it... Tomorrow the Dortmunder Flohmarkt has priority 

@Zarath :  The above was your comment almost a year ago, may I _kindly_ bring  this item to your attention again ?

redrathnure, 2025-01-01:
> perhaps one with a liteVNA can checkout https://github.com/zarath/nanovna-saver/tree/feature/clock_detect
> and do a debug output with nanovna-saver (-D debug.txt) and can attach the output on this discussion.

@zarath how to reach this branch?

redrathnure, 2025-01-21:
A https://github.com/NanoVNA-Saver/nanovna-saver/pull/754 was merged, now it should be possible to use some LiteVNA specific features (select data points up to 64K, whide band range, screenshots etc). Main functionality was tested with LiteVNA64, however it's just bare minimum. Feel free to check it with your device and open issue with bug or feature requests (e.g. [for this one](https://github.com/NanoVNA-Saver/nanovna-saver/pull/754#issuecomment-2604526261)).

Special thanks for @DiSlord for provided protocol description.