NanoVNA-Saver/nanovna-saver #541: 5.0.1 Crashes regularly during use, unexpected type 'float'

barjac, 2022-09-10:
I have just packaged 5.0.1 for Mageia and during testing I am seeing this repeated crash.

Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Frequency.py", line 448, in paintEvent
    self._check_frequency_boundaries(qp)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Frequency.py", line 463, in _check_frequency_boundaries
    qp.drawText(self.leftMargin + self.dim.width / 2 - 70,
TypeError: arguments did not match any overloaded call:
  drawText(self, Union[QPointF, QPoint], str): argument 1 has unexpected type 'float'
  drawText(self, QRectF, int, str): argument 1 has unexpected type 'float'
  drawText(self, QRect, int, str): argument 1 has unexpected type 'float'
  drawText(self, QRectF, str, option: QTextOption = QTextOption()): argument 1 has unexpected type 'float'
  drawText(self, QPoint, str): argument 1 has unexpected type 'float'
  drawText(self, int, int, int, int, int, str): argument 1 has unexpected type 'float'
  drawText(self, int, int, str): argument 1 has unexpected type 'float'
Aborted (core dumped)

Mageia Cauldron x86_64 

**Additional context**
Happens at various times which are not specifically reproduceable. Sometimes shortly after connecting to the device, other times when doing seemingly random things like checking a check box in settings.

barjac, 2022-09-12:
Thanks msspad - we now have another one..

Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Square.py", line 51, in paintEvent
    self.drawValues(qp)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Square.py", line 94, in drawValues
    self.drawMarker(x, y, qp, m.color, self.markers.index(m)+1)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Chart.py", line 310, in drawMarker
    cmarker.draw(x, y, color, f"{number}")
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Chart.py", line 93, in draw
    self.qp.drawText(x - text_width // 2, y - 3 - offset, text)
TypeError: arguments did not match any overloaded call:
  drawText(self, Union[QPointF, QPoint], str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRect, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, str, option: QTextOption = QTextOption()): argument 1 has unexpected type 'int'
  drawText(self, QPoint, str): argument 1 has unexpected type 'int'
  drawText(self, int, int, int, int, int, str): argument 2 has unexpected type 'float'
  drawText(self, int, int, str): argument 2 has unexpected type 'float'
Aborted (core dumped)

barjac, 2022-09-12:
I have now the following in 0.5.2 on program start:

Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Square.py", line 50, in paintEvent
    self.drawChart(qp)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Smith.py", line 101, in drawChart
    QtCore.QRect(center_x - 50, center_y - 4 + r, 100, 20),
TypeError: arguments did not match any overloaded call:
  QRect(): too many arguments
  QRect(int, int, int, int): argument 2 has unexpected type 'float'
  QRect(QPoint, QPoint): argument 1 has unexpected type 'int'
  QRect(QPoint, QSize): argument 1 has unexpected type 'int'
  QRect(QRect): argument 1 has unexpected type 'int'
Aborted (core dumped)

zarath, 2022-09-13:
Next version coming soon 🙈
Already found some other places where to cast to int.
Some libs seems to be pickier than before...

zarath, 2022-09-13:
hi @barjac can you check if this fixes that crashes?
https://github.com/zarath/nanovna-saver/tree/bugfix/541_crash_in_ri

barjac, 2022-09-13:
Looks identical to the error I reported yesterday :\
Crash on launch.

This package is from:
commit 2f69f5c (HEAD -> bugfix/541_crash_in_ri, origin/bugfix/541_crash_in_ri)

Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Square.py", line 50, in paintEvent
    self.drawChart(qp)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Smith.py", line 101, in drawChart
    QtCore.QRect(center_x - 50, center_y - 4 + r, 100, 20),
TypeError: arguments did not match any overloaded call:
  QRect(): too many arguments
  QRect(int, int, int, int): argument 2 has unexpected type 'float'
  QRect(QPoint, QPoint): argument 1 has unexpected type 'int'
  QRect(QPoint, QSize): argument 1 has unexpected type 'int'
  QRect(QRect): argument 1 has unexpected type 'int'
Aborted (core dumped)

barjac, 2022-09-13:
On 13/09/2022 17:48, Holger Müller wrote:
> hi @barjac can you check if this fixes that crashes?
> https://github.com/zarath/nanovna-saver/tree/bugfix/541_crash_in_ri
> 
> 
Hi Holger,
I replied in bug report.
Barry

zarath, 2022-09-15:
Casted r to int in Smith.py
can you test agian?

barjac, 2022-09-15:
[baz@localhost ~]$ nanovna-saver
NanoVNASaver 0.5.3-pre

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020ff NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Square.py", line 50, in paintEvent
    self.drawValues(qp)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Square.py", line 96, in drawValues
    self.drawMarker(x, y, qp, m.color, self.markers.index(m) + 1)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Chart.py", line 314, in drawMarker
    cmarker.draw(x, y, color, f"{number}")
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Chart.py", line 96, in draw
    self.qp.drawText(x - text_width // 2, y - 3 - offset, text)
TypeError: arguments did not match any overloaded call:
  drawText(self, Union[QPointF, QPoint], str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRect, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, str, option: QTextOption = QTextOption()): argument 1 has unexpected type 'int'
  drawText(self, QPoint, str): argument 1 has unexpected type 'int'
  drawText(self, int, int, int, int, int, str): argument 2 has unexpected type 'float'
  drawText(self, int, int, str): argument 2 has unexpected type 'float'
Aborted (core dumped)
[baz@localhost ~]$

zarath, 2022-09-15:
I've updated that branch again and hope to have now all floats forced to int for qp.draw.. methods.

barjac, 2022-09-15:
commit 3f8151aad72b9cb4155e85299f2d2f64f1b50570 (HEAD -> bugfix/541_crash_in_ri, origin/bugfix/541_crash_in_ri)
[baz@localhost ~]$ nanovna-saver
NanoVNASaver 0.5.3-pre

Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Square.py", line 50, in paintEvent
    self.drawValues(qp)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Square.py", line 96, in drawValues
    self.drawMarker(x, y, qp, m.color, self.markers.index(m) + 1)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Chart.py", line 315, in drawMarker
    cmarker.draw(x, y, color, f"{number}")
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Chart.py", line 96, in draw
    self.qp.drawText(x - int(text_width // 2),
TypeError: arguments did not match any overloaded call:
  drawText(self, Union[QPointF, QPoint], str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRect, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, str, option: QTextOption = QTextOption()): argument 1 has unexpected type 'int'
  drawText(self, QPoint, str): argument 1 has unexpected type 'int'
  drawText(self, int, int, int, int, int, str): argument 2 has unexpected type 'float'
  drawText(self, int, int, str): argument 2 has unexpected type 'float'
Aborted (core dumped)
[baz@localhost ~]$

I began to doubt myself when I saw the same errors as this morning, but the tarball I built from was made from that commit after a git pull on the same clone that I took earlier. Then built in mock under a new package name that I have checked is the one installed. I will re-build again later to be certain - just about to eat :)

zarath, 2022-09-15:
Yes. you are right!
With your data or layout a float glitched through for y value in marker iteration of square.py line 96...
https://github.com/zarath/nanovna-saver/commit/ef6a3c2d0a8f5b5a3865dd9bfd0aad286e5b7100

barjac, 2022-09-15:
NanoVNASaver 0.5.3-pre

Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Square.py", line 50, in paintEvent
    self.drawValues(qp)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Square.py", line 96, in drawValues
    self.drawMarker(x, y, qp, m.color, self.markers.index(m) + 1)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Chart.py", line 315, in drawMarker
    cmarker.draw(x, y, color, f"{number}")
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Chart.py", line 96, in draw
    self.qp.drawText(x - int(text_width // 2),
TypeError: arguments did not match any overloaded call:
  drawText(self, Union[QPointF, QPoint], str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRect, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, str, option: QTextOption = QTextOption()): argument 1 has unexpected type 'int'
  drawText(self, QPoint, str): argument 1 has unexpected type 'int'
  drawText(self, int, int, int, int, int, str): argument 2 has unexpected type 'float'
  drawText(self, int, int, str): argument 2 has unexpected type 'float'
Aborted (core dumped)

I am just re-building this to be sure I am at the right commit:
commit cabb8a43518969933a7ccc0664b7eb6e8cd9f4da (HEAD -> bugfix/541_crash_in_ri, origin/bugfix/541_crash_in_ri)
Author: Holger Müller <zarath@gmx.de>
Date:   Thu Sep 15 21:05:07 2022 +0200

    pycodestyle fixes

Testing again in a few moments...

barjac, 2022-09-15:
Seems OK now since that last commit. I now have graphs and no errors so far :)
I will play with it for a while and try to break it for you. :))

barjac, 2022-09-15:
Been playing with different antennas, bands etc. and no crashes.
The reference is not working, no reference traces remain or appear. The backgrounds to the graphs are black, so I changed the reference colour to cyan - still nothing, but not for this bug report.
While I have your attention :) I think that the sweep settings dialogue should close on hitting the "Set Band Sweep" button. I always expect it to do that but have to hit the window X as well.
Thanks for your efforts! I guess this can be closed now.

zarath, 2022-09-18:
Regarding UI behaviour.
Please make a fresh enhancement ticket..
I dunno when I'll look for such things. Currently I'm mostly focusing on refactoring of Analysis and Chart modules
to increase maintainability....

closed 2022-09-18 by zarath