NanoVNA-Saver/nanovna-saver #568: mutable default <class 'PyQt5.QtGui.QColor'>

omarfahmy, 2022-10-31:
Dear Sir, 
I am trying to run python file nanovna-saver.py. However, i got this error
Traceback (most recent call last):
  File "C:\Users\omarf\Desktop\Nano VNA\nanovna-saver-main\nanovna-saver.py", line 27, in <module>
    from NanoVNASaver.__main__ import main
  File "C:\Users\omarf\Desktop\Nano VNA\nanovna-saver-main\NanoVNASaver\__main__.py", line 36, in <module>
    from NanoVNASaver.NanoVNASaver import NanoVNASaver
  File "C:\Users\omarf\Desktop\Nano VNA\nanovna-saver-main\NanoVNASaver\NanoVNASaver.py", line 27, in <module>
    from NanoVNASaver import Defaults
  File "C:\Users\omarf\Desktop\Nano VNA\nanovna-saver-main\NanoVNASaver\Defaults.py", line 69, in <module>
    @DC.dataclass
     ^^^^^^^^^^^^
  File "C:\Python311\Lib\dataclasses.py", line 1221, in dataclass
    return wrap(cls)
           ^^^^^^^^^
  File "C:\Python311\Lib\dataclasses.py", line 1211, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\dataclasses.py", line 959, in _process_class
    cls_fields.append(_get_field(cls, name, type, kw_only))
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\dataclasses.py", line 816, in _get_field
    raise ValueError(f'mutable default {type(f.default)} for field '
ValueError: mutable default <class 'PyQt5.QtGui.QColor'> for field background is not allowed: use default_factory

Could you help me how to avoid such a problem?

Thanks & Best Regards

easytarget, 2022-12-10:
Possibly a Python 3.11 issue; I was having an identical issue after upgrading to Fedora37 (which has Python 3.11 as default), using Py 3.10 allowed the app to open.

On Fedora37 (eg Python 3.11 default):
```
$ mkdir VNA
$ cd VNA
$ sudo dnf install python3.10 python3.10-devel
$ python3.10 -m venv env10
$ source env10/bin/activate
(env10) $ python -m pip install --upgrade pip
(env10) $ python -m pip install wheel
(env10) $ python -m pip install https://github.com/NanoVNA-Saver/nanovna-saver/archive/refs/tags/v0.5.3.tar.gz
(env10) $ NanoVNASaver
NanoVNASaver 0.5.3
...
```
.. And the App opens.

zarath, 2022-12-10:
can you try latest sourcecode from master branch?

ewsandor, 2022-12-11:
I am also seeing this error after upgrading to Fedora 37.  

I tried pulling the tip of the main branch, but still see the same error with Python 3.11.  I can also confirm the steps shared by @easytarget to run nanovnasaver in a python 3.10 venv are working for me as well.

easytarget, 2022-12-12:
Unfortunately `main` also fails with Python 3.11 [on Fedora 37 (Gnome 43.2)]

```console
owen@sam VNA]$ rm -rf env*
[owen@sam VNA]$ python -m venv env11
[owen@sam VNA]$ source env11/bin/activate

(env11) [owen@sam VNA]$ python --version
Python 3.11.0

(env11) [owen@sam VNA]$ python -m pip install --upgrade pip wheel
Requirement already satisfied: pip in ./env11/lib64/python3.11/site-packages (22.2.2)
Collecting pip
  Using cached pip-22.3.1-py3-none-any.whl (2.1 MB)
Collecting wheel
  Using cached wheel-0.38.4-py3-none-any.whl (36 kB)
Installing collected packages: wheel, pip
  Attempting uninstall: pip
    Found existing installation: pip 22.2.2
    Uninstalling pip-22.2.2:
      Successfully uninstalled pip-22.2.2
Successfully installed pip-22.3.1 wheel-0.38.4

(env11) [owen@sam VNA]$ python -m pip install https://github.com/NanoVNA-Saver/nanovna-saver/archive/refs/heads/main.zip
Collecting https://github.com/NanoVNA-Saver/nanovna-saver/archive/refs/heads/main.zip
  Using cached https://github.com/NanoVNA-Saver/nanovna-saver/archive/refs/heads/main.zip
  Preparing metadata (setup.py) ... done
Requirement already satisfied: pyserial>=3.5 in ./env11/lib64/python3.11/site-packages (from NanoVNASaver==0.5.4rc0) (3.5)
Collecting PyQt5>=5.15.0
  Using cached PyQt5-5.15.7-cp37-abi3-manylinux1_x86_64.whl (8.4 MB)
Requirement already satisfied: numpy>=1.21.1 in ./env11/lib64/python3.11/site-packages (from NanoVNASaver==0.5.4rc0) (1.23.5)
Requirement already satisfied: scipy>=1.7.1 in ./env11/lib64/python3.11/site-packages (from NanoVNASaver==0.5.4rc0) (1.9.3)
Requirement already satisfied: Cython>=0.29.24 in ./env11/lib64/python3.11/site-packages (from NanoVNASaver==0.5.4rc0) (0.29.32)
Requirement already satisfied: PyQt5-sip<13,>=12.11 in ./env11/lib64/python3.11/site-packages (from PyQt5>=5.15.0->NanoVNASaver==0.5.4rc0) (12.11.0)
Requirement already satisfied: PyQt5-Qt5>=5.15.0 in ./env11/lib64/python3.11/site-packages (from PyQt5>=5.15.0->NanoVNASaver==0.5.4rc0) (5.15.2)
Building wheels for collected packages: NanoVNASaver
  Building wheel for NanoVNASaver (setup.py) ... done
  Created wheel for NanoVNASaver: filename=NanoVNASaver-0.5.4rc0-py3-none-any.whl size=203533 sha256=584eb542192c271437e6be36a554833b9054535e989114bc98dec6d7a29f43bb
  Stored in directory: /tmp/pip-ephem-wheel-cache-7d7ee8a9/wheels/58/01/0a/19c658220121a7ee7e85c7b658da55d4cc279957cb86ca7170
Successfully built NanoVNASaver
Installing collected packages: PyQt5, NanoVNASaver
Successfully installed NanoVNASaver-0.5.4rc0 PyQt5-5.15.7

(env11) [owen@sam VNA]$ NanoVNASaver 
Traceback (most recent call last):
  File "/home/owen/VNA/env11/bin/NanoVNASaver", line 5, in <module>
    from NanoVNASaver.__main__ import main
  File "/home/owen/VNA/env11/lib64/python3.11/site-packages/NanoVNASaver/__main__.py", line 36, in <module>
    from NanoVNASaver.NanoVNASaver import NanoVNASaver
  File "/home/owen/VNA/env11/lib64/python3.11/site-packages/NanoVNASaver/NanoVNASaver.py", line 27, in <module>
    from NanoVNASaver import Defaults
  File "/home/owen/VNA/env11/lib64/python3.11/site-packages/NanoVNASaver/Defaults.py", line 69, in <module>
    @DC.dataclass
     ^^^^^^^^^^^^
  File "/usr/lib64/python3.11/dataclasses.py", line 1221, in dataclass
    return wrap(cls)
           ^^^^^^^^^
  File "/usr/lib64/python3.11/dataclasses.py", line 1211, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/dataclasses.py", line 959, in _process_class
    cls_fields.append(_get_field(cls, name, type, kw_only))
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/dataclasses.py", line 816, in _get_field
    raise ValueError(f'mutable default {type(f.default)} for field '
ValueError: mutable default <class 'PyQt5.QtGui.QColor'> for field background is not allowed: use default_factory

# Failure...
# Now the same with Python 3.10

[... 33 lines cut ...]
Collecting PyQt5-Qt5>=5.15.0
  Using cached PyQt5_Qt5-5.15.2-py3-none-manylinux2014_x86_64.whl (59.9 MB)
Building wheels for collected packages: NanoVNASaver
  Building wheel for NanoVNASaver (setup.py) ... done
  Created wheel for NanoVNASaver: filename=NanoVNASaver-0.5.4rc0-py3-none-any.whl size=203533 sha256=f0949d5c135d6def7b84023b87188b588bd31f1008a9ce0f458fa3e1c4aa44b8
  Stored in directory: /tmp/pip-ephem-wheel-cache-s0dqzmd2/wheels/5b/c3/a9/6103212f379665ce007bd5a10d105ae4357148e920f4f8219f
Successfully built NanoVNASaver
Installing collected packages: pyserial, PyQt5-Qt5, PyQt5-sip, numpy, Cython, scipy, PyQt5, NanoVNASaver
Successfully installed Cython-0.29.32 NanoVNASaver-0.5.4rc0 PyQt5-5.15.7 PyQt5-Qt5-5.15.2 PyQt5-sip-12.11.0 numpy-1.23.5 pyserial-3.5 scipy-1.9.3

(env10) [owen@sam VNA]$ NanoVNASaver 
NanoVNASaver 0.5.4-pre

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020ff NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

# App works as normal.
```

easytarget, 2022-12-13:
Once back with my PC I'll try the fix from #572

easytarget, 2022-12-14:
Yes, the changes in #572 resolves this for me. 

I was able to start NanaVNASaver from a Python 3.11 venv and everything ran as expected (though I didn't go further than connecting my device and doing a basic sweep)

As a workaround you can create a venv and run: 
`python -m pip install https://github.com/zarath/nanovna-saver/archive/refs/heads/feature/python3_11.zip`

ewsandor, 2022-12-18:
Hi @zarath, I am also able to second @easytarget's observation.

juhanijaakola, 2022-12-20:
After applying changes in #572 I could start NanoVNASaver in Fedora 37 with python3-3.11.0-1.fc37. Thanks!

Please make an update to Fedora 37!

zarath, 2023-01-01:
fixed in #572

closed 2023-01-01 by zarath