NanoVNA-Saver/nanovna-saver #575: bug: Wont run on fedora 37

cacheflood, 2022-12-26:
Command line log:

NanoVNASaver 0.5.3

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020ff NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

(nanovna-saver:7384): GLib-GIO-ERROR **: 21:54:30.952: Settings schema 'org.gnome.settings-daemon.plugins.xsettings' does not contain a key named 'antialiasing'
Trace/breakpoint trap (core dumped)

easytarget, 2022-12-29:
See #568 and #572,
This is fixed in `main`, but needs releasing

The official fedora package is broken by this, you will need to wait for the maintainer of that to patch it.

LieBtrau, 2022-12-30:
FYI: This problem occurs also on Ubuntu 22.04.1 LTS, 64-bit, Wayland windowing system, Gnome version 42.5.

zarath, 2023-01-01:
Fixed in #572 / v0.5.4

closed 2023-01-01 by zarath