NanoVNA-Saver/nanovna-saver #578: Please add support for DeepVNA 101

yarodin, 2023-01-01:
Please add support for new hardware: Deepelec DeepVNA 101: https://deepelec.com/category/deepvna-101-en/
Current nanovna-saver detect it like ordinary NanoVNA which support datapoints: 11,51,101
But DeepVNA 101 in fact support datapoints: 11,51,101,201,301.
[image]
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/10329412/debug.txt)

zarath, 2023-01-09:
Do someone have a _debug.txt_ from an ordenary Nanovna-F, so I can distinguish them by info output?

yarodin, 2023-04-14:
debug.txt from ordinary Nanovna-F
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/11232585/debug.txt)

AdmiralAnimE, 2025-01-17:
Any progress on this? Additionally, screenshots have wrong colors.

[image]