NanoVNA-Saver/nanovna-saver #579: bug: CHANGELOG.md dropped by latest commit but still installed

yarda, 2023-01-03:
# Bug Report

**NanoVNA-Saver version:**
0.5.4

**Current behavior:**

```
running install_data
creating /builddir/build/BUILDROOT/nanovna-saver-0.5.4-1.fc38.x86_64/usr/share
creating /builddir/build/BUILDROOT/nanovna-saver-0.5.4-1.fc38.x86_64/usr/share/doc
creating /builddir/build/BUILDROOT/nanovna-saver-0.5.4-1.fc38.x86_64/usr/share/doc/nanovnasaver
error: can't copy 'CHANGELOG.md': doesn't exist or not a regular file
```

**Expected behavior:**

No error

**Steps to reproduce:**

```
python3 setup.py install
```

**Related code:**

The following patch should fix it (in case the CHANGELOG.md wasn't removed accidentally):

```
diff --git a/setup.py b/setup.py
index ca87cdf..29233ff 100644
--- a/setup.py
+++ b/setup.py
@@ -20,7 +20,7 @@ from setuptools import setup
 
 setup(
   data_files=[
-    ( "share/doc/nanovnasaver/", [ "CHANGELOG.md", "LICENSE", "README.md" ] ),
+    ( "share/doc/nanovnasaver/", [ "LICENSE", "README.md" ] ),
     ( "share/applications/", [ "NanoVNASaver.desktop" ] ),
     ( "share/icons/hicolor/48x48/apps/", [ "NanoVNASaver_48x48.png" ] ),
   ]
```

**Other information:**

yarda, 2023-01-03:
Dropped by commit: https://github.com/NanoVNA-Saver/nanovna-saver/commit/ed362a0c4bbd902434b4a7dd73bd2286a273a217

zarath, 2023-01-03:
Fixed in https://github.com/NanoVNA-Saver/nanovna-saver/commit/d654ea0441939e4e1c599d1333b587a185394fbe

closed 2023-01-03 by zarath