NanoVNA-Saver/nanovna-saver #586: bug: Crash while loading RSA3000 s2p file

baremetaldude, 2023-01-23:
# Bug Report

**NanoVNA-Saver version:**

0.5.4

**Current behavior:**

Crash

**Expected behavior:**

Displays sweep data

**Steps to reproduce:**

0. Use Rigol RSA3000 -> VNA mode -> S21 to generate s2p file
1. Files -> Load as sweep -> crash

**Other information:**
There are no S11 data in file.
Files -> Load as reference -> works! crashes only when loading as sweep data

[fb-smd-RSA3000.zip](https://github.com/NanoVNA-Saver/nanovna-saver/files/10479249/fb-smd-RSA3000.zip)

zarath, 2023-01-23:
It isn't reproducible here. Can you try starting nanovna-saver with the -D option and attach the output here?
nanovna-saver.py -D debug.txt

baremetaldude, 2023-01-23:
Probably it's TDR chart 

console log:
```
NanoVNASaver 0.5.4

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020ff NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

2023-01-24 01:22:53,623 - NanoVNASaver.Touchstone - WARNING - Comment after header: ! FREQ      S11REAL     S11IMAG     S21REAL     S21IMAG      S12REAL     S12IMAG     S22REAL     S22IMAG
NanoVNASaver\Charts\TDR.py:376: RuntimeWarning: invalid value encountered in scalar divide
Traceback (most recent call last):
  File "NanoVNASaver\Charts\TDR.py", line 375, in paintEvent
ValueError: cannot convert float NaN to integer
```

debug.txt:
```
2023-01-24 01:22:45,960 - NanoVNASaver - INFO - Startup...
2023-01-24 01:22:45,967 - NanoVNASaver.NanoVNASaver - DEBUG - Running from pyinstaller bundle
2023-01-24 01:22:45,972 - NanoVNASaver.NanoVNASaver - INFO - Settings from: C:/Users/jack/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
2023-01-24 01:22:45,972 - NanoVNASaver.Defaults - DEBUG - restored
(
CFG(gui=GUI(window_height=1344, window_width=2611, font_size=8, custom_colors=False, dark_mode=False, splitter_sizes=PyQt5.QtCore.QByteArray(b'\x00\x00\x00\xff\x00\x00\x00\x01\x00\x00\x00\x02\x00\x00\x01V\x00\x00\x03\xa2\x01\xff\xff\xff\xff\x01\x00\x00\x00\x01\x00'), markers_hidden=False), charts_selected=ChartsSelected(chart_00='S11 Smith Chart', chart_01='S11 Return Loss', chart_02='None', chart_10='S21 Polar Plot', chart_11='S21 Gain', chart_12='None'), chart=Chart(point_size=2, show_lines=True, line_thickness=1, marker_count=4, marker_label=False, marker_filled=False, marker_at_tip=False, marker_size=8, returnloss_is_positive=False, show_bands=False, vswr_lines=[]), chart_colors=ChartColors(background=<PyQt5.QtGui.QColor object at 0x000002A9E8BC70D0>, foreground=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7920>, reference=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7990>, reference_secondary=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7A00>, sweep=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7A70>, sweep_secondary=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7AE0>, swr=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7B50>, text=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7BC0>, bands=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7C30>), markers=Markers(active_labels=['actualfreq', 'impedance', 'serr', 'serl', 'serc', 'parr', 'parlc', 'vswr', 'returnloss', 's11q', 's11phase', 's21gain', 's21phase'], colored_names=True, color_0=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7450>, color_1=<PyQt5.QtGui.QColor object at 0x000002A9E8BC73E0>, color_2=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7370>, color_3=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7300>, color_4=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7290>, color_5=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7220>, color_6=<PyQt5.QtGui.QColor object at 0x000002A9E8BC71B0>, color_7=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7140>))
)
2023-01-24 01:22:45,972 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2023-01-24 01:22:45,972 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2023-01-24 01:22:45,972 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2023-01-24 01:22:45,972 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (3600000, 30000000)
2023-01-24 01:22:45,973 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2023-01-24 01:22:46,119 - NanoVNASaver.Hardware.Hardware - DEBUG - Interfaces: []
2023-01-24 01:22:46,120 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2023-01-24 01:22:46,149 - NanoVNASaver.Windows.SweepSettings - DEBUG - update_band(False)
2023-01-24 01:22:46,155 - NanoVNASaver.Defaults - DEBUG - storing
(
CFG(gui=GUI(window_height=1344, window_width=2611, font_size=8, custom_colors=False, dark_mode=False, splitter_sizes=PyQt5.QtCore.QByteArray(b'\x00\x00\x00\xff\x00\x00\x00\x01\x00\x00\x00\x02\x00\x00\x01V\x00\x00\x03\xa2\x01\xff\xff\xff\xff\x01\x00\x00\x00\x01\x00'), markers_hidden=False), charts_selected=ChartsSelected(chart_00='S11 Smith Chart', chart_01='S11 Return Loss', chart_02='None', chart_10='S21 Polar Plot', chart_11='S21 Gain', chart_12='None'), chart=Chart(point_size=2, show_lines=True, line_thickness=1, marker_count=4, marker_label=False, marker_filled=False, marker_at_tip=False, marker_size=8, returnloss_is_positive=False, show_bands=False, vswr_lines=[]), chart_colors=ChartColors(background=<PyQt5.QtGui.QColor object at 0x000002A9E8BC70D0>, foreground=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7920>, reference=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7990>, reference_secondary=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7A00>, sweep=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7A70>, sweep_secondary=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7AE0>, swr=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7B50>, text=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7BC0>, bands=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7C30>), markers=Markers(active_labels=['actualfreq', 'impedance', 'serr', 'serl', 'serc', 'parr', 'parlc', 'vswr', 'returnloss', 's11q', 's11phase', 's21gain', 's21phase'], colored_names=True, color_0=<PyQt5.QtGui.QColor object at 0x000002A9E8BC7450>, color_1=<PyQt5.QtGui.QColor object at 0x000002A9E8BC73E [... cut ...]
2023-01-24 01:22:46,157 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 165.000000, normal font: 165.000000, factor: 1.000000
2023-01-24 01:22:46,190 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2023-01-24 01:22:53,623 - NanoVNASaver.Touchstone - INFO - Attempting to open file T:/fb-smd-RSA3000.s2p
2023-01-24 01:22:53,623 - NanoVNASaver.Touchstone - INFO - !2-port S-parameter file
2023-01-24 01:22:53,623 - NanoVNASaver.Touchstone - WARNING - Comment after header: ! FREQ      S11REAL     S11IMAG     S21REAL     S21IMAG      S12REAL     S12IMAG     S22REAL     S22IMAG
```

closed 2023-02-24 by zarath