NanoVNA-Saver/nanovna-saver #589: Windows 11 continuous sweep does not stop with STOP button.

bearbate, 2023-01-28:
# Bug Report

**NanoVNA-Saver version:**

**Current behavior:**

**Expected behavior:**

**Steps to reproduce:**

**Related code:**

```
insert short code snippets here
```

**Other information:**

rjordans, 2023-02-12:
I can confirm this also happens on Linux

zarath, 2023-02-12:
See #128 
Any FW on any Device available which support disabling stimulus via command now?

rjordans, 2023-02-12:
@zarath this seems to be a different issue. looks like [this line](https://github.com/NanoVNA-Saver/nanovna-saver/blob/f996ee9ceb22cbf8e717e10ff4083dd03e93bfb9/NanoVNASaver/SweepWorker.py#L137) misses an "or self.stopped"

KJ7NLL, 2023-02-12:
+1, same issue, but we are in Linux running v0.5.4.  

if I open the "Sweep Settings" dialog and click it from "continuous" to "single" then it stops sweeping.  Then I put it back to "continuous" and click "sweep" to go back to continuous.  Ok for a work-around, but would be nice to click "Stop".

rjordans, 2023-03-31:
This should be fixed as of version 0.5.5 and can probably be closed now.  @bearbate and @KJ7NLL can you confirm this now works as expected?

closed 2024-06-26 by zarath