NanoVNA-Saver/nanovna-saver #594: doc: Linux - Add instructions for running without installation; add DEB / RPM build instructions

Ho-Ro, 2023-02-02:
The instructions omit the easiest way to get the program running under Linux - no installation - just start it in the git directory. This makes it difficult for pure users, e.g. hams, who therefore even try to run the Windows exe version under Wine.

Proposal - Add these sections below to the top README.md, e.g. between "Detailed installation instructions" and "Using the software" (Please review and add e.g. more necessary debian packages):

### Running on Linux without installation

The program simply works from the source directory without having to install it.

Simple step-by-step instruction, open a terminal window and type:
```sh
sudo apt install git python3-pyqt5 python3-numpy python3-scipy
git clone https://github.com/NanoVNA-Saver/nanovna-saver
cd nanovna-saver
```

Perhaps your system needs a few additional python modules:

- Run with `python nanovna-saver.py` and look at the response of (e.g. missing modules).
- Install the missing modules, preferably via `sudo apt install ...`

until `nanovna-saver.py` starts up.

Now the program can be used from the `nanovna-saver` directory.

### Installing via DEB for Debian (and Ubuntu)

The installation has the benefit that it allows you to run the program from anywhere, because the
main program is found via the regular `$PATH` and the modules are located in the Python module path.

If you're using a debian based distro you should consider to build your own `*.deb` package.
This has the advantage that NanoVNASaver can be installed and uninstalled cleanly in the system.

For this you need to install `python3-stdeb` - the module for converting Python code and modules into a Debian package:
```sh
apt install python3-stdeb
```
Then you can build the package via:
```sh
make deb
```
This package can be installed the usual way with
```sh
sudo dpkg -i nanovnasaver....deb
```
or
```sh
sudo apt install ./nanovnasaver....deb
```

### Installing via RPM (experimental)

`make rpm` builds an (untested) rpm package that can be installed on your system the usual way.

zarath, 2023-02-03:
Why bother cloning ;-)
`pip install pip install git+https://github.com/NanoVNA-Saver/nanovna-saver.git `
or
`pip install pip install git+https://github.com/NanoVNA-Saver/nanovna-saver.git@v0.5.4
`

Ho-Ro, 2023-02-03:
> Why bother cloning ;-)

Not only do I clone, I even fork many projects.
I can think of at least three reasons:
1. on a Debian system I prefer to install packages in deb format.
2. *I* want to have the current source code of *interesting* tools available on my PC. This also makes bug-hunting a lot easier. 
3. besides - when you were a child, didn't you also open your new technical devices no later than two days after Christmas? :)

KJ7NLL, 2023-02-12:
I agree that `make rpm` or `make deb` would be great.  Be sure to provide a way to specify the python interpreter path because Python 3.8 is pretty new and not all dists have that yet.

FYI, this procedure worked to build NanoVNA Saver on Oracle Linux 8 (RHEL8, etc): 

1. Download the FC36 RPM:
   - `wget  https://download-ib01.fedoraproject.org/pub/fedora/linux/updates/37/Everything/source/tree/Packages/n/nanovna-saver-0.5.4-1.fc37.src.rpm`
2. Build it with Python 3.8:
    1. rpm -ivh nanovna-saver-0.5.4-1.fc37.src.rpm
    2. Add this to the top of ~/rpmbuild/SPECS/nanovna-saver.spec:
        - `%define __python3 python3.8`
        - Replace BuildRequires packages with `python38` instead of `python3`:
			BuildRequires:  python38-setuptools
			BuildRequires:  python38-devel
			BuildRequires:  python3-pyserial
			BuildRequires:  python38-numpy
			BuildRequires:  python38-scipy
			BuildRequires:  python3-qt5
			BuildRequires:  python38-Cython
    3.  Build the package: `rpmbuild -bb ~/rpmbuild/SPECS/nanovna-saver.spec`
    4.  Install the result:
        - `rpm -ivh ~/rpmbuild/RPMS/noarch/nanovna-saver-0.5.4-1.el8.noarch.rpm`
3. Run it:
    1. Fix up user deps that didn't install:
        2. pip3.8  install --user --upgrade pip
        3. python3.8 -m pip install pyqt5
        4. python3.8 -m pip install pyserial
    2. Run it!  `NanoVNASaver`

KJ7NLL, 2023-02-12:
Btw, whoever handles `make rpm` buildout, you could start with `~/rpmbuild/SPECS/nanovna-saver.spec` as a template for your .spec (or .spec.in).

zarath, 2023-02-16:
> Not only do I clone, I even fork many projects. I can think of at least three reasons:
>
> 1. on a Debian system I prefer to install packages in deb format.
> 2. _I_ want to have the current source code [...]

I thought you've wanted pure user to have the easiest possible installation ;-)
Personally I almost always clone and build from source, whereas I use virtualenvs for python software to keep my system clean and being independent from dependencies of other python programs.
But nevertheless, all possibilities could be documented...

clytle374, 2025-04-15:
I find the instructions starting this post in docs/INSTALLATION.md

`nanovna-saver.py`
isn't in the main folder, but I find the following in src/NanoVNASaver.  When launching it I get the following error.  
`python src/NanoVNASaver/NanoVNASaver.py 

Traceback (most recent call last):
    File "/mnt/driveTwo/sources/nanovna-saver/src/NanoVNASaver/NanoVNASaver.py", line 28, in <module>
       from .About import VERSION

ImportError: attempted relative import with no known parent package
`
Is there an easy fix to this?  Can the instructions be updated?  Thanks, I'm running Gentoo and the available package is old.