NanoVNA-Saver/nanovna-saver #600: bug: crashing after connect - Rpi Bullseye

FilipCZE, 2023-02-13:
# Bug Report

Using Rpi3 with Bullseye verion of RpiOS.
After update NanoVNASaver to version 0.5.4 crashing again and again after connecting device.

**NanoVNA-Saver version:**

NanoVNASaver 0.5.4

**Current behavior:**

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020ff NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

libEGL warning: DRI2: failed to authenticate
2023-02-13 13:47:49,466 - NanoVNASaver.Hardware.NanoVNA_V2 - ERROR - Timeout reading version registers. Got: b''
2023-02-13 13:47:49,468 - NanoVNASaver.Controls.SerialControl - ERROR - Unable to connect to VNA: Timeout reading version registers
Traceback (most recent call last):
  File "/home/Filip/Install/nanovna-saver/NanoVNASaver/Controls/SerialControl.py", line 74, in serialButtonClick
    self.connect_device()
  File "/home/Filip/Install/nanovna-saver/NanoVNASaver/Controls/SerialControl.py", line 105, in connect_device
    frequencies = self.app.vna.readFrequencies()
  File "/home/Filip/Install/nanovna-saver/NanoVNASaver/Hardware/VNA.py", line 151, in readFrequencies
    return [int(f) for f in self.readValues("frequencies")]
  File "/home/Filip/Install/nanovna-saver/NanoVNASaver/Hardware/VNA.py", line 193, in readValues
    result = list(self.exec_command(value))
  File "/home/Filip/Install/nanovna-saver/NanoVNASaver/Hardware/VNA.py", line 96, in exec_command
    drain_serial(self.serial)
  File "/home/Filip/Install/nanovna-saver/NanoVNASaver/Hardware/Serial.py", line 33, in drain_serial
    cnt = len(serial_port.read(128))
  File "/home/Filip/.local/lib/python3.9/site-packages/serial/serialposix.py", line 560, in read
    raise PortNotOpenError()
serial.serialutil.PortNotOpenError: Attempting to use a port that is not open

**Expected behavior:**

not crahsing after connecting to device.

**Steps to reproduce:**

**Related code:**

```
insert short code snippets here
```

**Other information:**

closed 2023-06-15 by github-actions[bot]

FrankZyk, 2023-06-25:
Same crashing with version 0.5.5 on connect attempt, Win 10 64-bit executable download.

Ho-Ro, 2023-06-26:
> Same crashing with version 0.5.5 on connect attempt, Win 10 64-bit executable download.

I do not believe that your Win crash has exactly the same root cause as the original RPi crash. Please provide the logging output when started in a cmd.exe window.