NanoVNA-Saver/nanovna-saver #603: bug: Restoring saved 1-port calibration does not work

PetteriAimonen, 2023-02-21:
# Bug Report

**NanoVNA-Saver version:** Tested binary release 0.5.4 and git revision 09246b6a34530563f

**Current behavior:** Loading calibration from file works only for 2-port calibration.

**Expected behavior:** Loading should work for 1-port calibration also.

**Steps to reproduce:**
1. Perform 1-port calibration: Short, Open and Load. Apply calibration.
2. Save calibration to file.
3. Try to reload calibration from file.
4. Observe error messages `NanoVNASaver.Calibration - WARNING - Illegal data in cal file.` in console.

Saving and loading works fine for 2-port calibration (where Through, Thrurefl and Isolation are also included).

**Other information:**

I have attached saved 1-port and 2-port files.
The 1-port file only has the 1-port columns, which does not match the "long" format defined here: [Calibration.py](https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/NanoVNASaver/Calibration.py#L50-L60)
The header still has all labels, so it gets recognized as the long format.
[test_1port.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/10794974/test_1port.txt)
[test_2port.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/10794975/test_2port.txt)

PetteriAimonen, 2023-02-21:
As a workaround, one can manually edit the header to:

    # Hz ShortR ShortI OpenR OpenI LoadR LoadI ThroughR ThroughI IsolationR IsolationI

which corresponds to the "short" format.

Not sure whether the loading or the saving should be modified.

zarath, 2023-02-24:
Should be fixed with 0fbb301435f5a730ac1d85498a395dcaccf65a4e

PetteriAimonen, 2023-02-25:
Hmm, with a brief test it seems to work for 1-port but fail for 2-port now.

For me the saved headers are exactly equivalent for 1-port and 2-port cases, but the number of actual data columns is different (7 vs. 13)

zarath, 2023-03-01:
Loading and saving of Cal-Data is rewritten and more fault tolerant now.
can you test
https://github.com/NanoVNA-Saver/nanovna-saver/pull/607/commits/185a64b5ae51b4b21cb01ad4442fad8978993f8a
?

PetteriAimonen, 2023-03-01:
Works perfectly for me now, thanks a lot!

closed 2023-03-01 by zarath