NanoVNA-Saver/nanovna-saver #613: nanovna -saver crashes when NanoVNA-H4 connected

KG4IIC, 2023-03-09:
HP Pavillion x360 Convertable running linux Mint Kernel: 5.4.0-144-generic x86_64 bits: 64 compiler: gcc v: 9.4.0,  Desktop: Cinnamon 5.2.7 wm: muffin dm: LightDM Distro: Linux Mint 20.3 Una ,   base: Ubuntu 20.04 focal 

Followed the install instructions below:
      Running on Linux without installation
      The program simply works from the source directory without having to install it.
      
      Simple step-by-step instruction, open a terminal window and type:
      
      sudo apt install git python3-pyqt5 python3-numpy python3-scipy
      git clone https://github.com/NanoVNA-Saver/nanovna-saver
      cd nanovna-saver
      Perhaps your system needs a few additional python modules:
      
      Run with python nanovna-saver.py and look at the response of (e.g. missing modules).
      Install the missing modules, preferably via sudo apt install ...
      until nanovna-saver.py starts up.
      
      Now the program can be used from the nanovna-saver directory.

Without connecting the NanoVNA-H4,  nanovna-saver program opens and is responsive to input from mouse and keyboard.
When  NanoVNA-H4 is connected and then executing python3 nanovna-saver.py, it crashes at startup.
When  nanovna-saver program open and then  NanoVNA-H4 is connected, Refresh causes progrm to crash.
I have attached error message from terminal screenshot

[image]

KG4IIC, 2023-03-09:
[image]

NanoVNA-H4 specifications

zarath, 2023-03-10:
Is your linux user member of the dialout group?

closed 2023-08-04 by zarath