NanoVNA-Saver/nanovna-saver #637: bug:  Offset Delay of Through (in Calibration) is not working

k6jca, 2023-05-12:
# Bug Report

**NanoVNA-Saver version:**

Version 0.5.5

**Current behavior:**

Setting a Through standard's Offset Delay in the Calibration window has no effect.  For example, if the THRU's delay were set to 100 ps, S21 phase should change by -3.6 degrees every 100 MHz when measuring the THRU.  But phase does not change.  Instead it remains flat at 0 degrees.

This problem is caused by two misspellings in CalibrationSettings.py.  In each case the boolean cal_element.through_is_ideal in calculation() is misspelled as "throuh_is_ideal".

Correcting the spelling to "through_is_ideal" fixes the problem.

Code snippet:
   def calculate(self):
        cal_element = self.app.calibration.cal_element
        if self.app.sweep_control.btn_stop.isEnabled():
            self.app.showError(
                "Unable to apply calibration while a sweep is running."
                " Please stop the sweep and try again.")
            return

        cal_element.short_is_ideal = True
        cal_element.open_is_ideal = True
        cal_element.load_is_ideal = True
        cal_element.throuh_is_ideal = True  <-- Misspelled.  Change throuh to through

        # TODO: all ideal or not?
        if not self.use_ideal_values.isChecked():
            cal_element.short_is_ideal = False
            cal_element.open_is_ideal = False
            cal_element.load_is_ideal = False
            cal_element.throuh_is_ideal = False   <-- Misspelled.  Change throuh to through

zarath, 2023-05-13:
@k6jca thanks for the hint

closed 2023-05-13 by zarath

k6jca, 2023-05-13:
Many thanks, Holger.
- Jeff 

 
 

Closed #637 as completed.