NanoVNA-Saver/nanovna-saver #667: .exe not working

Chorbu, 2023-07-31:
Hello,

The last file (Windows Release #70: Commit [b480010]) doesn't start on Windows 11 64.

Dos windows starts and close but nothing else. The program doesn't start.

Best regards
Patrick

zarath, 2023-07-31:
I know. Therefore no binaries attached to the release files for windows.
I have no windows and I haven't found out, why pyinstaller doesn't catch all needed libraries for the windows executable.

Ho-Ro, 2023-08-01:
> Dos windows starts and close but nothing else. The program doesn't start.

Did you start the binary from a cmd.exe window? If not, please do so and report any messages.

t52ta6ek, 2023-08-01:
Greetings! I was able to get 0.6.1 to build and run under Win10 x86_64 without problems with my project directly cloned from this latest repository, no changes.

I built using the Git Bash environment (MinGW64) on Win10, with Python 3.11.4 and a Python virtual enviroment pretty much set up following the process in the Installation Instructions for Ubuntu for nanovna-saver. Everything below was performed in the Git Bash shell in Win10.

cd into cloned nanovna-server directory, then start the virtual environment:
. ~/.venv_nano/Scripts/activate

(After cloning, I reinstall the various components.)
pip install -U pip setuptools setuptools_scm wheel auto-py-to-exe -e .

(Then, confirm the program runs by typing: NanoVNASaver in the Git Bash window.)

Next, build the Windows .exe using either auto-py-to-exe or pyinstaller. The pyinstaller command line created by auto-py-to-exe was:
pyinstaller --noconfirm --onefile --console --icon "C:/Users/t52ta6ek/Desktop/work/nanovna-saver/icon_48x48.ico"  "C:/Users/t52ta6ek/Desktop/work/nanovna-saver/nanovna-saver.py"

Using the command line above, the .exe lands in the .../nanovna-saver/dist directory. I then just copy the .exe to my Win10 desktop and done.  I did notice some weirdness with the sweep crashing, but after deleting the existing .ini files nanovna-saver created down in the AppData\Roaming, the sweep started working again. Not sure what was going on there and the crash hasn't re-occured - knock-on-wood.

Git Bash - https://gitforwindows.org/

zarath, 2023-08-01:
Regarding sweep crash. There was a problem with the sweep.py getter/setter changes - fixed in 0.6.2.
Current problem in building for windows in github actions is that pyinstaller didn't find/bundle the qt libs in the build env, resulting in an 5MB binary which is obviously not containing all needed data.

zarath, 2023-08-03:
Python version for windows was to old *bummer*

closed 2023-08-03 by zarath