NanoVNA-Saver/nanovna-saver #670: feat: Create Windows executable with this project's icon rather than a generic python application icon

t52ta6ek, 2023-08-03:
# Feature Request

Allow github action release_win.yml to create a nanovna-saver.exe for Windows containing this project's icon, rather than creating the .exe with a generic python application icon.

**Describe Preferred Solution**

**Describe Alternatives**

**Related Code**

**Additional Context**

**If the feature request is approved, would you be willing to submit a PR?**
_(Help can be provided if you need assistance submitting a PR)_

- [X ] Yes
- [ ] No

zarath, 2023-08-03:
Duplicate of #667

zarath, 2023-08-03:
I would be happy if binary creation of windows exe in github actions would work again...

closed 2023-08-03 by zarath

t52ta6ek, 2023-08-03:
Good luck in tracking that down. I had to look up when you meant by "github actions" and just found out how you *really* build the release executable images. :-)

Learning quite a bit myself just contributing what little I can here (and hopefully without breaking anything!) HAHAHA

Thanks again and thanks for maintaining this project.

Best!!!
Tom