NanoVNA-Saver/nanovna-saver #675: Arm64 build with memory limits.

jackokring, 2023-08-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Yes, it sure is at a freeze when doing an arm64 build and somehow over 2.0GB is used and er, locked up.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I'd like a single core efficient on memory build option.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I've aliased make ... blah, blah.

**Additional context**
Add any other context or screenshots about the feature request here.

I'm not interested in an x86 build.

zarath, 2023-08-08:
I guess target OS should be OpenBSD, right?

closed 2023-08-11 by zarath

jackokring, 2023-08-20:
Chromebook, debian container.

dothebart, 2023-08-22:
most probably the required devices will not be visible within the container anyways.