NanoVNA-Saver/nanovna-saver #676: Controlling device via Serial commands

krupis, 2023-10-17:
# Feature Request

Hello. I am looking for potential solutions for our problem. We need to be able to perform RF tests on 2.4Ghz antenna. One of the possible solutions is to use NanoVNA device. 

**Describe Preferred Solution**

In order to perform all the tests we want, we need to be able to control the device using serial commands. Using serial commands, we need to do the following:

- Set frequency range, span, center frequency;
- Set maximum measurement points;
- Place cursors on specific frequencies;
- Read dB measurement data from set cursors.

We do not need any GUI , we only want to send the command and receive a simple response in text. Do you know if anything similar is currently possible?

**Additional Context**

Could you also clarify to me what kind of NanoVNA devices do you support? From what I understand, there are quite a few different options such as:
NanoVNA H
NanoVNA H4
NanoVNA V2 (design by HCXQS/NanoRF)
NanoVNA V3 (design by HCXQS/NanoRF)

and probably a few others that I dont know about.

Does your tool support all versions of NanoVNA?

krupis, 2023-10-18:
Thanks for your quick reply. I have now realized that you are the same person that have responded on the:
https://groups.io/g/nanovna-users/topic/102014357

suggesting me to use https://github.com/Ho-Ro/nanovna-tools

Could you please help me understand what are the differences between these two tools and which one is more suitable for my particular application?

> Please be more specific what kind of tests, S11 with the Antenna alone or S21 with a calibrated reference antenna?
We will need to do both of those tests.

I currently have my hands on NanoVNA -H (I am aware that it is not capable of measuring 2.4 GHz but I can use this to familiarise myself with the tools available and how to use them). I can then later buy the NanoVNA V2 and I expect I would be able to use the tools in a Similar way.

Ho-Ro, 2023-10-18:
For a 1st start you can try also with the -H, it can go up to 2.7 GHz - but with degraded performance because the oscillator chip can only deliver frequencies up to 300 MHz and the device uses the harmonics of the generated square wave for the higher frequency ranges. 
About your question which tool to use, I would go with an easy terminal based approach, connect the -H to USB and try to open a terminal program, e.g. hterm or putty and play with the commands. Press <ENTER> and get a response, if this works, type help <ENTER> and see the commands, etc.
Then try the same with "nanovna_command.py" from my toolbox. If this works, you can use this program or "nanovna_snp.py" as a template for own experiments.

krupis, 2023-10-18:
@Ho-Ro 
Understood. I have already managed to get the https://github.com/Ho-Ro/nanovna-tools up and running

I have managed to get the nanovna_command.py working. For example I can execute sweep command:
```
C:\Users\petrikas.lu\Downloads\nanovna-tools-main\nanovna-tools-main>python nanovna_command.py sweep
10000000 1500000000 101
```

However, I am not fully understanding how to use ```nanovna_capture.py``` and ```nanovna_snp.py``` as I am getting the following errors:
```
C:\Users\petrikas.lu\Downloads\nanovna-tools-main\nanovna-tools-main>python nanovna_snp.py
# HZ S RI R 50
Traceback (most recent call last):
  File "C:\Users\petrikas.lu\Downloads\nanovna-tools-main\nanovna-tools-main\nanovna_snp.py", line 174, in <module>
    output_string( format_parameter_line( line ) )
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\petrikas.lu\Downloads\nanovna-tools-main\nanovna-tools-main\nanovna_snp.py", line 129, in format_parameter_line
    freq, S11r, S11i = line.split()
    ^^^^^^^^^^^^^^^^
ValueError: too many values to unpack (expected 3)
```

and
```
C:\Users\petrikas.lu\Downloads\nanovna-tools-main\nanovna-tools-main>python nanovna_capture.py
Traceback (most recent call last):
  File "C:\Users\petrikas.lu\Downloads\nanovna-tools-main\nanovna-tools-main\nanovna_capture.py", line 115, in <module>
    if bytestream == b'capture?\r\nch> ': # error message
       ^^^^^^^^^^
NameError: name 'bytestream' is not defined
```

Ho-Ro, 2023-10-18:
> NameError: name 'bytestream' is not defined

This is an error, sorry, the line must be 
```python
if captured_bytes == b'capture?\r\nch> ': # error message
```

but nevertheless you're getting an error in communication (Windows?, see below)
But I would propose to get the nanovna_command.py running, then you have the core communication template available. The capture program captures a screenshot of the device and relies on a perfect low level communication because it transfers the complete screen content in one rush without much protocol overhead. That means on the other hand that comm error spoils the complete transfer.

1st step: try with a terminal and make you familiar with the ASCII communication
2nd step: try the python tools

My development system is Linux, Windows behaves probably different. You can put print(...) statements before the lines in trouble, e.g. at the end of `nanovna_command.py` and see the format, adapt the program etc.

```python
response = echo[ :-len( crlf + prompt ) ]             # remove '\r\nch> '

print( response ) # Ho-Ro DEBUG

if outfile == sys.stdout:
    print( response.decode() )                        # write string to stdout
else:
    outfile.write( response + lf )                    # write bytes to outfile```

krupis, 2023-10-18:
@Ho-Ro 

I do not fully understand what you mean:

> 1st step: try with a terminal and make you familiar with the ASCII communication

That is exactly what I am doing, no? Using various python commands such as ```nanovna_command``` and ```nanovna_capture``` and others using terminal? I do not know what else could you be refering to when you say ```try with a terminal```

krupis, 2023-10-18:
I understand you now.. You refer to using serial commands directly via ```serial terminal```. I use ```Termite```:
[image]

I will familiarise myself with this. Thanks!

Ho-Ro, 2023-10-18:
The cammand format is described [here](https://groups.io/g/nanovna-users/wiki/12760) and [there](https://groups.io/g/nanovna-users/files/NanoVNA_Console_Commands_Dec%209-19.pdf). Otherwise gives `help` an overview which commands are available.

Ho-Ro, 2023-10-18:
Please follow the discussion at the right repo:
https://github.com/Ho-Ro/nanovna-tools/discussions/2