NanoVNA-Saver/nanovna-saver #684: dev: is cython used at run time? at all?

asarhaddon, 2023-12-16:
Hello.
Cython is listed in `setup.cfg` and `requirements.txt`.
Automated tools building the Debian package deduce the run-time dependencies from them.
Cython does not seem to be used at all, but even if it is, it is most probably only useful at build time.
Could it be removed from these lists and handled separately, like the dependencies for tests?
Thanks.

hosiet, 2025-01-23:
The mentioning of Cython has been completely dropped in https://github.com/NanoVNA-Saver/nanovna-saver/commit/588c8548add8f99bdc67c57a2df8c7bb66f9870a#diff-50c86b7ed8ac2cf95bd48334961bf0530cdc77b5a56f852c5c61b89d735fd711 . This change has not been reflected in a released version yet.

Please consider closing this issue once nanovna-save makes another release.

asarhaddon, 2025-01-29:
We may even close the issue now if that fits your workflow. A patch already removes cython from the Debian package.
Also, next release will not be packaged any time soon because uv is quite complex to package properly.

closed 2025-02-07 by zarath