NanoVNA-Saver/nanovna-saver #686: feat: 0.6.3 - why no binary?

myhillman, 2024-01-10:
# Feature Request

**Describe the Feature Request**

Not all users, in fact the vast majority I suspect, are not able to build an executable. A binary should be a feature of every release

**Describe Preferred Solution**

Provide a binary

ajawamnet, 2024-01-17:
yea really... too man-bun java jockey for me.  Include a binary.  I have work to do... not futzing around with your source tree.

zarath, 2024-01-18:
> yea really... too man-bun java jockey for me. Include a binary. I have work to do... not futzing around with your source tree.
Then write your own Software and don't expect other people to write it for you in theirs leisure time!

closed 2024-01-18 by zarath

ajawamnet, 2024-01-18:
If it wasn't for us hardware guys you wouldn't have anything to write software on...