NanoVNA-Saver/nanovna-saver #688: bug: NanoVNASaver won't start on macOS

MKesenheimer, 2024-01-15:
# Bug Report

**NanoVNA-Saver version:**
00dd59ffc68a37b60cf25ffb2574d4f32dca1e06

**Current behavior:**
`NanoVNASaver` does not start and throws the following error when started from the command line:
```
[90198] Module object for pyimod02_importers is NULL!
Traceback (most recent call last):
  File "PyInstaller/loader/pyimod02_importers.py", line 22, in <module>
  File "pathlib.py", line 14, in <module>
  File "urllib/parse.py", line 40, in <module>
ModuleNotFoundError: No module named 'ipaddress'
Traceback (most recent call last):
  File "PyInstaller/loader/pyiboot01_bootstrap.py", line 17, in <module>
ModuleNotFoundError: No module named 'pyimod02_importers'
[90198] Failed to execute script 'pyiboot01_bootstrap' due to unhandled exception: No module named 'pyimod02_importers'
[90198] Traceback:
(null)
```

**Expected behavior:**
`NanoVNASaver` starts.

**Steps to reproduce:**
Start the `NanoVNASaver` application on macOS (Apple M1 Pro, Sonoma 14.2.1 (23C71), Python 3.11.7).

```
$ python --version
Python 3.11.7
$ ./dist/NanoVNASaver/NanoVNASaver
```

lagomorph, 2024-03-25:
Not the exact same issue but build-macos-app.sh no longer works with python 3.12. Upgrading pip and doing the steps in the script manually gives an executable that crashes on startup.

```
$ python --version
Python 3.12.2
$ ./dist/NanoVNASaver/NanoVNASaver
setuptools_scm/_integration/setuptools.py:30: RuntimeWarning: 
ERROR: setuptools==0.dev0+unknown is used in combination with setuptools_scm>=8.x

Your build configuration is incomplete and previously worked by accident!
setuptools_scm requires setuptools>=61

Suggested workaround if applicable:
 - migrating from the deprecated setup_requires mechanism to pep517/518
   and using a pyproject.toml to declare build dependencies
   which are reliably pre-installed before running the build tools

NanoVNASaver 0.6.3.post1.dev2+ga04d6d9

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020ff NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

zsh: segmentation fault  ./dist/NanoVNASaver/NanoVNASaver
```

treaves, 2024-11-09:
Too bad you've received no response. Have either of you tried recently?  I just cloned & built and the app works for me, other that some not-uncommon layout issues I may help them resolve.

phikes, 2025-03-23:
Possibly related: https://github.com/NanoVNA-Saver/nanovna-saver/issues/803#issuecomment-2746150058