NanoVNA-Saver/nanovna-saver #689: NO Binari...bug:

KN4USA, 2024-01-26:
# Would be nice to have the install file available for Windows but instead the page reads "No Binari" and has no install files available.

<

pgramsey, 2024-01-28:
I've been wondering what's happening with 0.6.3. Is Windows no longer going to be supported?

dovandung, 2024-03-03:
I move directory src\NanoVNASaver up same level with nanovna-saver.py and remove two line :
# Ignore the current working directory.
#src = os.path.join(os.path.dirname(__file__), "src")

#if os.path.exists(src):
#    sys.path.insert(0, src)
after that I build exe by pyinstaller , it works

ChuckMarshall, 2024-03-22:
Can you provide a link to download windows 64 NanoVNA Saver?

zodoczi, 2024-03-27:
Yes, a Windows binary would be really helpful! Thank you for the great work!

dovandung, 2024-06-24:
https://drive.google.com/file/d/1a2zpwb791Ho-Xe7uuZbT3t8KHf5rGN5X/view?usp=drive_link

zarath, 2024-06-25:
V0.6.4 has a binary again

KN4USA, 2024-06-25:
Stop emailing me. I have nothing to do with this

zarath, 2024-06-26:
> Stop emailing me. I have nothing to do with this
> […](#)

The fix your github settings or don't participate in issues here.

closed 2024-06-26 by zarath

dothebart, 2024-06-26:
I'm just a taxi driver.