NanoVNA-Saver/nanovna-saver #691: Support for 201 sweep points when using NanoVNA-F V2

haupte, 2024-02-10:
# Feature Request

I still suffer from the same problem as described by alfagulf in #371 3 years ago:

> Firmware V0.2.0 for the NanoVNA-F V2 now support upto 201 sweep points.
> 
> The NanoVNA Saver V0.3.8 program throws an exception when the NanoVNA-F V2 with firmware V0.2.0 is set to a number of sweep points greater than the default 101.
> 
> ```
> 2021-01-26 12:51:15,147 - NanoVNASaver.SweepWorker - ERROR - list assignment index out of range
> Traceback (most recent call last):
>   File "/home/sabah/Documents/Electronics/NanoVNA-F V2/nanovna-saver-master/NanoVNASaver/SweepWorker.py", line 80, in run
>     self._run()
>   File "/home/sabah/Documents/Electronics/NanoVNA-F V2/nanovna-saver-master/NanoVNASaver/SweepWorker.py", line 124, in _run
>     self.updateData(freq, values11, values21, i)
>   File "/home/sabah/Documents/Electronics/NanoVNA-F V2/nanovna-saver-master/NanoVNASaver/SweepWorker.py", line 178, in updateData
>     self.data11[offset + i] = data11[i]
> IndexError: list assignment index out of range
> ```
> 
> I thought I bring this to your attention, hoping you rectify this in your new release.
> 
> Thanks Sabah

- My hardware type is SysJoint NanoVNA-F V2.
- My firmware release is v0.5.0. This release allows for setting the device to 201 sweep points maximum.
- I use vnasaver v0.6.2.post1.dev2+g2f8c534 (because I don't know how to compile release 0.6.3) that detects the hardware as "NanoVNA-F_V2" and its firmware as "NanoVNA-F_V2 v0.5.0".

When I connect vnasaver to NanoVNA-F V2 then the NanoVNA is remotely set to 101 sweep points although before I'd  set it to 201 sweep points manually. In vnasaver there are fixed options to set 11, 51 and 101 sweep points and "Custom points" allow 101 points maximum.

Wouldn't there be a solution to raise the maximum number of sweep point in vnasaver?

Tnx, Erich

alfagulf, 2024-02-10:
Hello Haupte,
This is what I found 3 years ago.

I thought I share with you my solution to the fact that the NanoVNS Saver program V0.3.8 lists only 11,51,101 datapoint in the settings although the NanoVNA-F V2 new firmware allows upto 201 sweep data points.
I found that under the Hardware folder there are python files for each supported hardware, in the NanoVNA-F_V2.py file I add the line shown in green to the class section as follows:

class NanoVNA_F_V2(NanoVNA):
    name = "NanoVNA-F_V2"
    screenwidth = 800
    screenheight = 480
    valid_datapoints = (101, 11, 51, 201)

I hope that may be of a help to some of you.
Best regards,

haupte, 2024-02-10:
Thank you, dear Sabah!

I tried the same source code modification but, unfortunately, my knowledge is not sufficient to do an on-line build targetting Windows. For now, the only possible pre-defined options with "make" are Linux variants.

I will have to wait until Windows builds are standard again. 😅

I will keep the issue open. Perhaps there is an official developer who shows mercy to me and releases a Windows execution module.

Sincerly, 
Erich

xros, 2025-01-19:
Hi boys, @haupte @alfagulf @lx1cw I think this has been added in my pull-request. https://github.com/NanoVNA-Saver/nanovna-saver/pull/760 . I can't believe this issue has been their for 4 years😳  You guys can try and test😊!

Previously I added support for NanoVNA F V3 whose max datapoints can reach up to 801 from here https://github.com/NanoVNA-Saver/nanovna-saver/issues/711 . NanoVNA F V2 is also a cool hardware though it's not opensource hardware which violates some licenses.

superduty, 2025-03-26:
I have a Sysjoint NanoVNA-F_V2 with 0.5.4 firmware.  I downloaded your release last week and it works well.  When I use the 0.7.3 version of from the original repository (https://github.com/NanoVNA-Saver/nanovna-saver), the sweep point selecton box only goes to  101.  If I check the custom box, I won't let me enter anything higher than 101.

So, I cloned that repository just for kicks to see why.  I am not at all familiar with GIT and Python is not my language.  It seems that the tool for building was changed at some point between your version NanoVNASaver-v0.6.9.dev13-x86_64-Windows and the current 0.7.3 and now is using UV.  After a few days I finally found how to build with uv on Windows using git-bash and my local clone.  I had to make two changes after lots of logger.debug statements I found that the function "get_features" in the Hardware/NanoVNA_F_V2.py file was missing a "return self.features" .  Also, I had to add the "init_features" function to the same file in order to get the SN to show in the info box.  I copied this function from the F_V3 file.  I now have a Windows 10 exe file that works for the Sysjoint V2 device and is updated to 0.7.3 changes.

I created an Issue #38 for the missing return statement.  I have never created a Git fork or done a pull.  I am reading the GIT documents and the proper procedure to do a fork and submit my changes but I am a little hesitant for fear of not doing it properly.