NanoVNA-Saver/nanovna-saver #711: Support NanoVNA-F V3 & more than 101 sweep points.

OP-drone, 2024-08-08:
SysJoint NanoVNA-F V3 fw 0.5.3 support up to 801 sweep points.

NanoVNA-Saver v0.6.4 there are  fixed options to set 11, 51, 101 sweep points and "Custom points" 101 points maximum.

Need support more than 101 sweep points.

xros, 2024-12-26:
Hopefully this can be supported. I purchased a NanoVNA-F v3 with fw 0.5.8

BTW, do you know where to download the latest firmware for nanovna f v3?

patdoss, 2024-12-30:
Hi,
I purchased also the SysJoint NanoVNA-F V3 HW: Rev D and FW 0.5.8 two weeks ago and already done some code modifications yesterday, desesperating to be limited to 101 sweep points for a so nice toy, so i started the debug mode on the latest release NanoVNA-Saver v0.6.9.
As a first approach, the code update to support the 801 sweep points is fast and easy to do.
For early testing here is what i done:
- add a new file under /src/NanoVNASaver/Harware/NanoVNA_F_V3_.py
- add small modifications to /src/NanoVNASaver/Harware/Hardware.py
  >> from NanoVNASaver.Hardware.NanoVNA_F_V3 import NanoVNA_F_V3
        ...
  >> NAME2DEVICE = {
       ....
       "F_V3": NanoVNA_F_V3,
       ..
It's only for early testing for NanoVNA-F V3 frustrated owners like me and i probably miss many other details.. i just start to learn the tool !
Thanks so much to the developpers for this beatifull tool.

[Hardware.zip](https://github.com/user-attachments/files/18278322/Hardware.zip)
[NanoVNA_F_V3_.zip](https://github.com/user-attachments/files/18278289/NanoVNA_F_V3_.zip)
[image]

patdoss, 2025-01-04:
Hi,
I don't know yet if it is used elsewhere in the code, but I manage to get the SN display in the device configuration form.
I found the problem in the /src/NanoVNASaver/Harware/VNA.py

It's not the "sn:" string to detect BUT "SN:" with my hardware so i added a "OR"

   def read_features(self):
       .....
       .....
        #mod#
        if "sn:" or "SN:" in result:
            self.features.add("SN")
            self.SN = self.getSerialNumber()
        #end-mod#

Also at the end of the file, the command to send is "SN" and not "sn" (i just commented the line ans replace the command name as i don't know if other hardwares can use both "sn" or "SN".. )

    def getSerialNumber(self) -> str:
        # return " ".join(list(self.exec_command("sn")))
        return " ".join(list(self.exec_command("SN")))
        
Sorry if my info is not at the right place or not usefull but if i can help... :-)
Thank's a lot.
73's F6EXL

[image]

[VNA.zip](https://github.com/user-attachments/files/18305188/VNA.zip)

xros, 2025-01-08:
I have made a pull-request and tested with my own NanoVNA F V3 to prove that it works with this modification from here: https://github.com/NanoVNA-Saver/nanovna-saver/pull/751

Hopefully this functionality can be added.  Special thanks to @patdoss 

Before merging to the main branch, the fully runnable version is in my branch: https://github.com/xros/nanovna-saver/tree/sysjoint . Anyone can download and test with their own NanoVNA and NanoVNA F V3.

[image]

xros, 2025-01-19:
This feature has already been merged based on the pull-request https://github.com/NanoVNA-Saver/nanovna-saver/pull/751 

Everyone can test their NanoVNA-F V3 from the **main** branch of this project now.

You are welcome.

patdoss, 2025-01-19:
Thanks a lot ! 73's Patrick

closed 2025-02-09 by OP-drone