NanoVNA-Saver/nanovna-saver #712: nanovna-saver.py cant start because it cant find its own version information.

takigama, 2024-08-12:
**Describe the bug**
I've been trying to get nanovna-saver running on Linux inside a Chromebook and kept getting stuck with this traceback:
```
Traceback (most recent call last):
  File "/tmp/nanovna-saver/./src/NanoVNASaver/About.py", line 22, in <module>
    version = get_version(root='..', relative_to=__file__)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/setuptools_scm/__init__.py", line 148, in get_version
    _version_missing(config)
  File "/usr/lib/python3/dist-packages/setuptools_scm/__init__.py", line 108, in _version_missing
    raise LookupError(
LookupError: setuptools-scm was unable to detect version for /tmp/nanovna-saver/src.

Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.

For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/tmp/nanovna-saver/./nanovna-saver.py", line 32, in <module>
    import NanoVNASaver.__main__
  File "/tmp/nanovna-saver/./src/NanoVNASaver/__main__.py", line 33, in <module>
    from NanoVNASaver.About import version, INFO
  File "/tmp/nanovna-saver/./src/NanoVNASaver/About.py", line 24, in <module>
    from NanoVNASaver._version import version
ModuleNotFoundError: No module named 'NanoVNASaver._version'
```

**To Reproduce**
Steps to reproduce the behavior:
1. get a Chromebook (or Chrome OS flex, or similar, they run debian bookworm (12), not sure if that would be sufficient though for reproducing the issue)
2. git clone the repo
3. install the required packages (there's a few)
4. cd into the repo and run ./nanovna-saver.py

**Expected behavior**
nanovna-saver to start

**Desktop (please complete the following information):**
 - OS: ChromeOS Linux (debian 12)
 - Python version:3.11.2
 - NanoVNA-Saver version: from git

**Additional context**
There is a simple work around, you can just edit nanovna-saver/./src/NanoVNASaver/About.py and hard-code the version to something, for eg:

```
from setuptools_scm import get_version
try:
    #version = get_version(root='..', relative_to=__file__)
    version = "0.6.6"
```

and nanovna-saver will start up just fine. There are few search results to be found with the error, but they all seem to stem from an issue in git that doesn't seem to apply here.

running it with "SETUPTOOLS_SCM_DEBUG=1 ./nanovna-saver.py" wasn't hugely helpful either.

EnPassant123, 2024-09-28:
The correct fix is the following:

Replace
`version = get_version(root='..', relative_to=__file__)`
with 
`version = get_version(root='../..', relative_to=__file__)`

This is because the directory with the About.py folder is two layers above the directory with the .git folder.

zarath, 2024-10-18:
merged in main branch

closed 2024-10-18 by zarath