NanoVNA-Saver/nanovna-saver #718: macOS build for latest

fred-corp, 2024-10-14:
Will there be a macOS build for the latest release ?

If not, are there instructions to compile from source ?

Thanks !

aviatorhh, 2024-10-16:
There is a [build-macos-app.sh](https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/build-macos-app.sh) in the root dir.

closed 2024-10-16 by fred-corp

rrobinett, 2024-10-17:
ON my Mac Mw running Sequoia 15.0.1:

./build-macos-app.sh         runs with warnings

but it doesn't run:

Robs-MacBook-Air-M2:nanovna-saver rob$ ./nanovna-saver.py
Traceback (most recent call last):
  File "/Users/rob/ham/nanovna-saver/./nanovna-saver.py", line 32, in <module>
    import NanoVNASaver.__main__
  File "/Users/rob/ham/nanovna-saver/./src/NanoVNASaver/__main__.py", line 31, in <module>
    from PyQt6 import QtWidgets
ModuleNotFoundError: No module named 'PyQt6'
Robs-MacBook-Air-M2:nanovna-saver rob$

fred-corp, 2024-10-17:
I couldn't run the build script without errors either, but I managed to install the app with MacPorts ([instructions here](https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/docs/INSTALLATION.md))

rrobinett, 2024-10-18:
Thanks!  The MacPorts solved my problem