NanoVNA-Saver/nanovna-saver #733: bug:

naiiawah, 2024-12-05:
# Bug Report

**NanoVNA-Saver version:**

v0.6.5 + a few commits (SHA b375d8e) on Mac Ventura v13.7.1 Intel

**Current behavior:** 

App crashes after pressing the connect to device button.

**Expected behavior:** 

Connects to device without crashing.

**Steps to reproduce:**

Build SHA b375d8e and execute the app.  Press the connect to device button, immediately crashes. Build the previous SHA and no crash.

**Related code:**

```
insert short code snippets here
```

**Other information:**

naiiawah, 2024-12-05:
I can supply the MacOS Problem Report (crash report) if needed.

zarath, 2024-12-05:
Console output of during the crash would be nice. Preferably with -D debug.txt as startup option. Files can be uploaded to issues...

naiiawah, 2024-12-05:
Here is the console output:

> brianm@Cosmos:~/Sources/Ham/nanovna-saver/dist/NanoVNASaver.app/Contents/MacOS ((HEAD detached at b375d8e))
> $ ./NanoVNASaver -D debug.txt
> NanoVNASaver 0.6.5.post1.dev8+gb375d8e
> 
> Copyright (C) 2019, 2020 Rune B. Broberg
> Copyright (C) 2020ff NanoVNA-Saver Authors
> 
> This program comes with ABSOLUTELY NO WARRANTY
> This program is licensed under the GNU General Public License version 3
> 
> See https://github.com/NanoVNA-Saver/nanovna-saver for further details.
> 
> Traceback (most recent call last):
>   File "NanoVNASaver/NanoVNASaver.py", line 569, in dataUpdated
>   File "NanoVNASaver/Windows/TDR.py", line 196, in updateTDR
> UnboundLocalError: cannot access local variable 'td' where it is not associated with a value
> Abort trap: 6
> brianm@Cosmos:~/Sources/Ham/nanovna-saver/dist/NanoVNASaver.app/Contents/MacOS ((HEAD detached at b375d8e))
> $ ls
> NanoVNASaver debug.txt

I've attached the "debug.txt" file.

[debug.txt](https://github.com/user-attachments/files/18030555/debug.txt)

EnPassant123, 2024-12-06:
I think I found it.

Someone moved the logic for lowpass TDR to this function:
def _tdr_lowpass(self, tdr_format, s11, windowed_s11)

and the td variable is no longer in the same scope as the updateTDR function, which is causing the error.

naiiawah, 2024-12-06:
That was my conclusion as well after I looked at the diffs of the TDR file between the SHA's.  Just wasn't sure of the right way to fix it.

closed 2024-12-06 by zarath

naiiawah, 2024-12-06:
Looks like the fix in merge #734 got eclipsed away by merge #735.

zarath, 2024-12-06:
It was my fault, as I wanted to make the linter happy ... so I split out that function, not a aware of the scope of td data.

naiiawah, 2024-12-07:
Just to be clear, this bug needs to be re-opened.  It is broken on HEAD again.

commit e7cf6b17 (2024-12-06): fix: #733 crash in TDR