NanoVNA-Saver/nanovna-saver #744: bug: Python dependency versions  too tight for package maintenance

barjac, 2024-12-31:
# Bug Report

**NanoVNA-Saver version:**

0.6.8

Packages cannot be installed because the distro (Mageia  10 dev) already has much newer versions of some deps which are specified in requirements.txt and pyproject.toml with '==' rather than '>='.

Can these requirements allow for newer installed versions that are still compatible?

Packages should run as well as build :)

I will list the currently troublesome deps on a different machine later.

barjac, 2025-01-01:
Current nanovna-saver requires             Current Mageia 10 (dev) versions

```
python(abi)[== 3.12]                       3.12.8 OK
python3.12dist(cython)[== 3.0.11]          3.0.11 OK
python3.12dist(numpy)[== 2.1.3]            2.1.3  OK for now but update available
python3.12dist(pyqt6)[== 6.7.1]            6.7.1  OK
python3.12dist(pyqt6-sip)[== 13.8]         13.9.1    <<<<
python3.12dist(pyserial)[== 3.5]           3.5.5  OK
python3.12dist(scipy)[== 1.14.1]           1.14.1 OK
python3.12dist(sip)[== 6.8.6]              6.9.1     <<<<
```
Can these ( <<<< ) be relaxed to use our later versions?

zarath, 2025-01-21:
changed with switching from setuptools to uv