NanoVNA-Saver/nanovna-saver #747: dev: What about Poetry

redrathnure, 2025-01-01:
Hi, what do you think about switching build and packaging to [Poetry ](https://python-poetry.org/) tool? 

A few reasons why it may be better:
1. All dependencies are in the same place, including dev and something which is declared as mandatory for result application. Now this information is spread around multiple files (setup.cfg, project.toml, requirements.tx, GH actions).
2. All project/package meta information in one file 
3. Better version control of dependencies. Most likely related to https://github.com/NanoVNA-Saver/nanovna-saver/issues/744
4. Match more easier and consistent development environment. E.g. unified way to manage venvs, install dependencies (prod and dev scopes), building and publishing result artifacts and more
5. Unified (and simpler) way to build and publishing result python artifacts
6. Better IDE support

... and more of what a modern build and package system has to offer

zarath, 2025-01-05:
I think it will convert to poetry in future... Luckily python 3.8 is EOL and there was no whining why the software didn't work on extremely ancient machines anymore... But I need time and desire to clean things up.

redrathnure, 2025-01-05:
@zarath We actively use it so I may help with it. And I would recommend try UV instead. It's much faster and seems cover all use cases.

I may prepare PR (of two), at least as start point for discussion. WDYT?

> Luckily python 3.8 is EOL 

BTW Looks like the (real) minimal python version here is 3.10. Some of libs does not support 3.8 and 3.9 anymore

closed 2025-01-13 by redrathnure