NanoVNA-Saver/nanovna-saver #759: bug: ModuleNotFoundError: No module named 'NanoVNASaver._version' when compiling using the uv python package mananger

xros, 2025-01-19:
# Bug Report

**NanoVNA-Saver version:**
main branch at the time of writting

**Current behavior:**

### ModuleNotFoundError: No module named 'NanoVNASaver._version'

Before we used `pip install -r requirments.txt` to install default packages needed for nanovna-saver, but @redrathnure  changed to package manager **uv** https://github.com/astral-sh/uv  from this pull-request: https://github.com/NanoVNA-Saver/nanovna-saver/pull/752

**I don't use uv to create virtual enviroment, because I need 2 defined virtual enviroment for building binaries for both ARM64 macOS and X86_64 macOS**

In a pre-configured python virtual enviroment for macOS ARM64, and another virtual environment for macOS x86_64

`uv pip sync pyproject.toml` It automatically installed some packages in my **defined** virtual enviroment, but when I started to test the app using `python nanovna-saver.py` it had this bug
```
  File "/Users/someone/joy/X_proj/nanovna-saver-dev/nanovna-saver/src/NanoVNASaver/app_version.py", line 9, in <module>
    from ._version import version
ModuleNotFoundError: No module named 'NanoVNASaver._version'
```

I read the doc <https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/docs/DEVELOPMENT.md>, and tried to manually install the packages by switching back to **pip** by doing this `uv pip compile pyproject.toml -o requirements.txt`. This will create the file **requirements.txt** back. Then I tried with `pip install -r requirements.txt`, then tried to re-run the app `python nanovna-saver.py` and it had the same error.

The error `ModuleNotFoundError: No module named 'NanoVNASaver._version'` pointed to this file  <https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/src/NanoVNASaver/app_version.py#L9>. I think it's because I didn't use **uv** to create a virtual enviroment in `.env` folder and other auto-machines were stopped.

I used this method I wrote here for building apps <https://github.com/xros/nanovna-saver/wiki/Making-apps-for-MacOS> for different architectures of macOS including ARM64 and x86_64 on 1 Apple Computer. 

**By using 2 different pre-defined virtual enviroments, I don't need to re-install all the packages again everytime in the ONLY 1 folder .env (which uv creates) for us, and it's fast when building apps**

**Expected behavior:**

`python nanovna-saver.py` should run without error

**Steps to reproduce:**

As above 

**Related code:**

file <https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/src/NanoVNASaver/app_version.py#L9>

```python
import importlib.metadata

APP_VERSION = "unknown"
try:
    # Change here if project is renamed and does not equal the package name
    APP_VERSION = importlib.metadata.version("NanoVNASaver")
except importlib.metadata.PackageNotFoundError:  # pragma: no cover
    # Looks like we neded this case for apps out of pyinstaller packages
    from ._version import version

    APP_VERSION = version
```

**Other information:**

Can someone come up with a method which can keep that **using pip as package manager is still available** for some cases and not using **uv** for creating the python virtual environment?

redrathnure, 2025-01-20:
_version is generated by setuptools_scm package and it is not related to venv or package installation. You may run 'python -m setuptools_scm' inside venv or check documentation.

About requerements management, 'uv export' may be used to form requirements.txt file, then you may use "usual" pip/pip3 to prepare venv "manually". 'uv pip instal' may also work and be more faster option, however I am not sure how it manage multiarch configurations.

xros, 2025-01-20:
> _version.py is generated by setuptools_scm package and it is not related to venv or package installation. You may run 'python -m setuptools_scm' inside venv or check documentation.
> 
> About requerements management, 'uv export' may be used to form requirements.txt file, then you may use "usual" pip/pip3 to prepare venv "manually". 'uv pip instal' may also work and be more faster option, however I am not sure how it manage multiarch configurations.

Exactly. `uv pip install` works fine if you only build for 1 arch.  the file _version.py can be auto-generated. But if I want to build for multiarch on 1 computer, the `venv` virtual environment should be replaced too. I'm finding a method to alternate the virtual environment for multiarch

redrathnure, 2025-01-20:
https://docs.astral.sh/uv/concepts/projects/config/#project-environment-path may be helpful for this case. Seems it posdible to change location or venv using env vars. However I still not sure how uv handles multiplatform builds (e. g. will it choose right wheel pakage to install)

xros, 2025-01-20:
> https://docs.astral.sh/uv/concepts/projects/config/#project-environment-path may be helpful for this case. Seems it posdible to change location or venv using env vars. However I still not sure how uv handles multiplatform builds (e. g. will it choose right wheel pakage to install)

Currently I delete the `.venv` manually for different architectures everytime. <https://github.com/xros/nanovna-saver/wiki/Making-apps-for-MacOS#1-manually-remove-the-env-enviroment-everytime-you-change-architecture>

But the `uv` project claimed **there's not a way to define multiple persistent environments.** from https://github.com/astral-sh/uv/issues/7778

redrathnure, 2025-01-20:
Question/idea: do you need to use UV to build application package at all? 

In theory it should be possible to build python package by `uv build`, which is cross platform. And then install it into the empty/multi-arch/special venv (just `pip install package_file` command). This should install nanovna-saver together with all runtime dependencies without need to mess with dev dependencies and all related tools. And then it should be possible to run pyinstaller in the prepared/multi-arch venv. It scans all dependencies anyway, so should not be big difference between `raw sources` from src dir and installed package (which at the end the same sources but in different location).

@xros could you try it? Would it simplify macos package building process?

xros, 2025-01-20:
Right now I think the work-around would be that find a method to automatically change the `.venv` when using `uv`. 

Such as 

```bash
# build 1 (on arm64)
# do something in shell, change to arm64 shell
uv sync
uv build xxxxx
# build 2 (on x86_64)
# do something in shell, change to x86_64 shell
mv .env .env_arm64
uv sync
uv build xxxxx
# build 3 (on arm64)
# then switch to another architecture, do something in shell, change to arm64 shell
mv .env .env_x86_64
ln -sf .env_arm64 .env
uv sync
uv build xxxxx
```

redrathnure, 2025-01-20:
I still not sure why do you need to swap dirs manually. E.g. following code should work if you need different venv:
```
export UV_PROJECT_ENVIRONMENT=.env_arm64
#uv sync && uv build
# or just
uv build

export UV_PROJECT_ENVIRONMENT=.env_x86_64
#uv sync && uv build
# or just
uv build
```
More details is [here](https://docs.astral.sh/uv/concepts/projects/config/#project-environment-path)

> And if uv still can't come up with a method to deal with packaging for different architectures. It won't be very useful as pip.

Just FYI uv is **not** for building. pip is **not** for building too. This project use setuptools (*.gz and wheel) to prepare python packages and pyinstall to prepare binary distributions.

Plus keep in mind, the project does not have any platform specific libs or native bindings, and result package has a valid meta information. So you may build source package, move it to any platform and prepare venv with whatever tool do you prefer. This means you do **not** need any `uv' or 'uv pip` at all, you do not need to rebuild the package (`uv build`) per each arch nether. Just prepare right venv, install the package (the rest will be installed as dependency) and run pyinstaller.

xros, 2025-01-21:
I think because you didn't try, that's why you don't know why. 

If you have M chip mac computer by side, you can try.

Here's the error that `uv` returns
```text
(env_313_x86_64_macos) MacBookPro:nanovna-saver $ mv .venv .venv_x86_64
(env_313_x86_64_macos) MacBookPro:nanovna-saver $ export UV_PROJECT_ENVIRONMENT=.venv_x86_64
(env_313_x86_64_macos) MacBookPro:nanovna-saver $ uv run NanoVNASaver -d
warning: `VIRTUAL_ENV=/Users/someone/nanovna-saver-dev/env_313_x86_64_macos` does not match the project environment path `.venv_x86_64` and will be ignored
error: Failed to spawn: `NanoVNASaver`
  Caused by: No such file or directory (os error 2)
```
I thought that's the error from system parameters. So I changed to the full path 
```export UV_PROJECT_ENVIRONMENT=`pwd`"/.venv_x86_64"``` but it still has the same error **does not match the project environment path `.venv_x86_64` and will be ignored**

As I shared with you from here <https://github.com/NanoVNA-Saver/nanovna-saver/issues/759#issuecomment-2601200183>

> > https://docs.astral.sh/uv/concepts/projects/config/#project-environment-path may be helpful for this case. Seems it posdible to change location or venv using env vars. However I still not sure how uv handles multiplatform builds (e. g. will it choose right wheel pakage to install)
> 
> Currently I delete the `.venv` manually for different architectures everytime. https://github.com/xros/nanovna-saver/wiki/Making-apps-for-MacOS#1-manually-remove-the-env-enviroment-everytime-you-change-architecture
> 
> But the `uv` project claimed **there's not a way to define multiple persistent environments.** from [astral-sh/uv#7778](https://github.com/astral-sh/uv/issues/7778)
> 
> The `uv` binaries are in different architectures. And if I don't want to pollute my original Python enviroment, I would have to manually firstly create a virtual enviroment A, then use the A to install `uv` binaries, then with `uv` create another virtual enviroment under folder `.env` B which is unpleasant.

From `uv` developers: **there's not a way to define multiple persistent environments.** 

The do-able method I found is to either delete the `.venv` or move it somewhere and create a softlink to it, like what I did above.

----
FYI, I know `uv` is not for building, maybe there's some misunderstandings here. I use pyinstaller for building apps from commands in terminal sometimes or some CI. I saw your previous commit which removed the file `requirements.txt` entirely which made me feel odd. They can exist together.

Certainly we can get the file `requirements.txt` back by `uv pip compile pyproject.toml -o requirements.txt`.

Here I address that **if you want to get requirements.txt, then you'll have to install virtual enviroment A, then in A install uv. Then uv will create another virtual enviroment B for development purposes.** Of cause we don't need to use `uv`, but where is the `requirements.txt` in the repo now? 

I think `uv` is cool but we can still have something traditional as well.

For buildings app for mutiple architectures on different OS, the simpler the better.

xros, 2025-01-21:
Just as I tried
> From uv developers: **there's not a way to define multiple persistent environments.**
> The do-able method I found is to either delete the .venv or move it somewhere and create a softlink to it, like what I did above.

I tried once again. `UV_PROJECT_ENVIRONMENT` has nothing to do with virtual environment. when you run applications. `uv run NanoVNASaver -d`, it ONLY looks for folder `.venv`! And the `uv` develpers won't care till now.

redrathnure, 2025-01-21:
```
> uname -a
Darwin xxx 24.2.0 Darwin Kernel Version 24.2.0: Fri Dec  6 19:02:41 PST 2024; root:xnu-11215.61.5~2/RELEASE_ARM64_T6030 arm64

> uv --version
uv 0.5.10 (37b11ddb2 2024-12-17)

> uv run whereis python
python: /tmp/nanovna-saver/.venv/bin/python

> export UV_PROJECT_ENVIRONMENT=".test"

> uv run whereis python

...
Creating virtual environment at: .test
Installed 48 packages in 228ms
python: /tmp/nanovna-saver/.test/bin/python

#  Cleaning up
> rm -rf .env
> rm -rf .test

> ls .venv
ls: .venv: No such file or directory

> ls .test
ls: .test: No such file or directory

# Trigger venv preparation
> uv sync

# Nothin in default venv location
> ls .venv
ls: .venv: No such file or directory

# However it respects UV_PROJECT_ENVIRONMENT=".test" settings
> ls .test
CACHEDIR.TAG bin          lib          pyvenv.cfg

# Try to run app
> uv run NanoVNASaver

# Wait till App window open, then close it

# Check venv again

# Nothin in default venv location
> ls .venv
ls: .venv: No such file or directory

# However it respects UV_PROJECT_ENVIRONMENT=".test" settings
> ls .test
CACHEDIR.TAG bin          lib          pyvenv.cfg
```
Sorry, I have not idea why it does not works in your case.

redrathnure, 2025-01-21:
And, just as illustration for "uv and requriements.txt less" approach:

```
> rm -rf dist

> uv build

> ls dist
NanoVNASaver-0.6.8.post1.dev3+ge3587e2.d20250121-py3-none-any.whl nanovnasaver-0.6.8.post1.dev3+ge3587e2.d20250121.tar.gz

# BTW, I have no idea why wheel package has so weird naming

# Prepare new venv manually.
# In case of multyarch build this step may be more complex 
> mkvirtualenv test_2
....

# No need any dev packages, no need of uv export, just install nanovnasaver package
> pip install dist/nanovnasaver-*.tar.gz
....

> pip list
Package      Version
------------ -----------------------------------
NanoVNASaver 0.6.8.post1.dev3+ge3587e2.d20250121
numpy        2.2.2
pip          24.3.1
PyQt6        6.8.0
PyQt6-Qt6    6.8.1
PyQt6_sip    13.9.1
pyserial     3.5
scipy        1.15.1
setuptools   75.6.0
wheel        0.45.1

# Just install pyinstaller and you're good to go
```