NanoVNA-Saver/nanovna-saver #768: Enhanced response in LiteVNA64

PA3CCE, 2025-01-21:
# Feature Request
Please make activating the "Enhanched Response" feature of LiteVNA64 in NanoVNA-Saver possible.

"Enhaced Response" is a feature under "Calibration" . 
It corrects in the  S21 calibration ,  the impedance  misfit  of  S21  to  50Ohms.  
This is necessay only, when  S21 measurements are to be executed.

DiSlord, 2025-01-24:
ER calibration always used in NanoVNA Saver by default

PA3CCE, 2025-01-26:
ER calibration always used in NanoVNA Saver by default
====================================

Is it ?
 In the latest distributed executable the lLiteVNA still is recognised as "SAA2" .

Does  SAA2 have this option ??

redrathnure, 2025-01-26:
> In the latest distributed executable the lLiteVNA still is recognised as "SAA2" .

Most likely LiteVNA changes were not released yet (no new builds on[ the release page](https://github.com/NanoVNA-Saver/nanovna-saver/releases)). @PA3CCE  could you run app from the latest master state and check how LiteVNA is recognized?

redrathnure, 2025-01-27:
OK, I see.I published a pre-build a package from the latest branch: https://drive.google.com/file/d/1sLRYhjEchmYgg0-jmMWxvFdN6N3_grbY/view?usp=sharing, this may be simpler that install python and other dev tools.

PA3CCE, 2025-01-28:
When I have the  timeI,  I do some more tests on NanoVNA-Saver with my LiteVNA64, I found that the font can be set upto included 12, that's great!
I found also  that when I start the program I have to set mij privious settings again,  I personally I would like that the last setiings are  saved when I exit the program.
Whle measuring, I saw that VNA's screen was flashing between USB mode and the standard measuring screen.
I will continue testing  and let you know my findings.
Please tell me if I need to install more Python libraries, and if so which ones,  to be possible to build NanoVNA-Saver myself from your latest updated sources,
Thank you.

redrathnure, 2025-01-28:
First of all thanks for testing and feedback.

About LiteVNA

> Whle measuring, I saw that VNA's screen was flashing between USB mode and the standard measuring screen.

Currently this is "default mode" for the LiteVNA. NanoVNA-Saver tries to block device only when it's needed, mostly for reading data. 
Please note that settings and calibration data are **not** shared between device and NanoVNA-Saver. This means you may have different sweep and display settings (and should have different calibrations) for the device and app on the same time. 
I thought it would be useful e.g. to have wide range and small amount of measurement point on device (having kind of overview) and narrow range with higher amount of points and avg in app to have more precise data. Otherwise just work with NanoVNA-Saver and do not pay attention to device:)
However if switching between USB and normal modes is a problem, we may disable this feature or may do it configurable.

> I found also that when I start the program I have to set mij privious settings again, I personally I would like that the last setiings are saved when I exit the program.

Just paid attention I have similar issue. Maybe some regressing from latest changes. Would be nice if you could open a bug for settings saving/restoring. Would easier to not forget about this point.

> Please tell me if I need to install more Python libraries, and if so which ones, to be possible to build NanoVNA-Saver myself from your latest updated sources,

OK, first of all https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/docs/DEVELOPMENT.md document may be interested for you. 
In short, you need to perform 3 main steps: prepare minimal local environment, install project and run it (or build win exec file)

About "prepare minimal local environment" step: you would need a [Python](https://www.python.org/downloads/windows/) and [Astral UV](https://docs.astral.sh/uv/getting-started) tool. Actually there are two options here:
1. Install UV (e.g. `powershell -ExecutionPolicy ByPass -c "irm https://astral.sh/uv/install.ps1 | iex"` or `cargo install --git https://github.com/astral-sh/uv uv` or `scoop install main/uv`, please see https://docs.astral.sh/uv/getting-started/installation/#standalone-installer for more details) and this tool will itself install Python, venv and all required libraries.
2. Install [Python](https://www.python.org/downloads/windows/), then install  uv (`pip install uv` or `pipx install uv`). 

I would say option 1 is easier for case when you start configuration from scratch. Option #2 is better if you already have some Python or related tools installed.

Then you would need to "build" project, which includes a) preparing venv b) installing required libs and c) do some application preparation. Hopefully all this can be done via one shingle command `uv run task ui-compile` run from root of the project.

Then you have 2 options to run application:
1. `uv run NanoVNASaver` if you what to just lunch the  application "with the latest changes"
2. `uv run task build-pkg-win` if you what to build windows exe file, e.g. to copy it to different location/machine or share with somebody.

BTW, if you have any ideas how to improve https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/docs/DEVELOPMENT.md documentation, e.g.  to be more understandable, please share them. Perhaps the project may benefit from good development documentation (easier start for beta testers, contributors etc).

PA3CCE, 2025-01-29:
==However if switching between USB and normal modes is a problem, we may disable this feature or may do it configurable.

Not a real problem for me, I wiil just neglect it.

==Would be nice if you could open a bug for settings saving/restoring. Would easier to not forget about this point.

Will do so immediately..

== #2 is better if you already have some Python or related tools installed.

That's what I did.

==Then you would need to "build" project, which includes a) preparing venv b) installing required libs and c) do some application ==preparation. Hopefully all this can be done via one shingle command uv run task ui-compile run from root of the project.

OK, fine !

Then you have 2 options to run application:

Graet !!

Thank you so much for all info !!

DiSlord, 2025-01-29:
Register V2/Lite
0x26 - (byte) mode (0 - USB mode, 1 - send raw data, 2 - normal mode, 3 - send calibrated data)

Send if need not calibrated data
20 26 00

Send if need calibrated data
20 26 03

Send if need leave USB mode
20 26 02

DiSlord, 2025-01-29:
my firmware V2/LiteVNA Binary protocol:
V2 contain 256 registers for read/write

On write to registers data put to reg[address] and V2 grab data and process (depend from address)

All values after in hex format!!!!
Commands for read/write
0x0D - read version (return "2")
0x10 - read byte
0x11 - read word
0x12 - read dword
0x13 - read qword

0x18 - read measured data (use 30 address for count)

0x20 - write byte
0x21 - write word
0x22 - write dword
0x23 - write qword
0x28 - write buffer (next byte data size, and data)

address:
0x00 - (qword) start frequency
0x10 - (qword) frequency step
0x20 - (word) points count
0x22 - (word) values pre frequency (need set to 1)

0x26 - (byte) mode (0 - USB mode, 1 - send raw data, 2 - normal mode, 3 - send calibrated data)
0x30 - (byte) get data points count (bytw limit to 255, but for Lite possible set to 0 for get all measured points from point count)

0x40 - (byte) average
0x41 - (byte) si power
0x42 - (byte) adf power
0x44 - (byte) write channel select (1 - send only CH0 data, 2 send only CH1, any other - send both)

0x50 - (dword) color value
0x54 - (byte) color index (use NanoVNA H/H4 indexes)
0x58 - (dword) write set time in UNIX format
0x5C - (word) read battery value in millivolts

0xEE - (byte) any value, on write V2 send screenshot

0xD0 - (dword) SN[0] (serial number 0)
0xD4 - (dword) SN[1] (serial number 1)
0xD8 - (dword) SN[2] (serial number 2)

0xF0 - (byte) device variant (2 for V2)
0xF1 - (byte) protocol version (1 for V2)
0xF2 - (byte) board revision (2 for V2, 3 - V2Plus, 4 - V2Plus4)
0xF3 - (byte) firmware major (1 for all)
0xF4 - (byte) firmware minor

Example:
Set "average = 5" command:
20 - write byte
40 - address (average)
05 - value
Need send : 204005

Set color[7]=RRGGBB:
20 - write byte
54 - address (color index)
07 - value (7 index is LCD_TRACE_2_COLOR)
22 - write dword
50 - address (color value)
BB - color in RGB
GG - color in RGB
RR - color in RGB
00 -
Need send: 2054072250BBGGRR00

Read color from color[8]:
20 - write byte
54 - address (color index)
08 - value (8 index is LCD_TRACE_3_COLOR)
12 - read dword
50 - address (color value)
Need send: 2054081250
rx *00 fc 00 00
BB GG RR 00

Example read data:
11.382 tx: setting scan parameters
20 30 00 - (set point count as 0, all points)
23 00 10 27 00 00 00 00 00 00  - write start = 10000
23 10 C7 03 00 00 00 00 00 00  - write stop  = 1000000
21 20 00 04 - points count 0x400 = 1024
21 22 01 00 - values per freq = 1

11.391 tx: requesting 1024 points ..
11.395 tx: 
20 30 00 - obsolete?
18 30 FF - read 255
18 30 FF - read 255
18 30 FF - read 255
18 30 FF - read 255
18 30 04 - read 4

redrathnure, 2025-01-29:
@DiSlord 

> my firmware V2/LiteVNA Binary protocol:

How to recognize your firmware from "something from factory" one? And do the protocols different for the both devices (v2 and LiteVNA) or diff relevant only for V2?

DiSlord, 2025-01-30:
Official LiteVNA firmware based on my stable version firmware, so protocol equal

PA3CCE, 2025-02-03:
@ Maxim Medvedev :

I was busy doing other things, sorry.

As said before, I installed "UV" on my system.
Today I additionally installed the packages "Dependencies" and  "Configsettings" on my computer.
Now, when I run "uv run NanoVNA-Saver-Main"  in the command window,  using the latest sources, I get the following

Quote:

Error : Failed to generate package metadata for : 'nanovnasaver==0.6.8.post1.dev19+gcead804.d20250201 @ editable+."
  caused by : The build backend returned an error
  Caused by : 'Call to 'setuptools_wrapper.build_editable' failed (editcode 1)

[stderr]
Traceback (most recent call last):
 File <"string"> line 14, in <module
    requires = get requires_for_build({})
 File "C:\.......AppData\local\uv\cache\buids-v0.templefwpf\lib\site-packages\setuptools\build_meta.py", line 483 in        
   get_requires_for_build_edtable
    return self.get_requires_for_get_wheel(config_settings)
                                         ^^^^^^^^^^^^^^^^^
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

### etc etc etc, ending up with : 
 
LookupError: Setuptools.scm was unable to detect version for D:\...\...\ nanovna-saver-main.
 
Unquote.

I'm so sorry to bother you again abt this, what is it that I'm missing more ???
Thank you in advance.

redrathnure, 2025-02-03:
I would try `uv run NanoVNASaver `. Not sure uv understands  "uv run NanoVNA-Saver-Main" .

PA3CCE, 2025-02-03:
I would try uv run NanoVNASaver . Not sure uv understands "uv run NanoVNA-Saver-Main" .
=================

I tried, same result.
Anyhow, thanks for the response.

redrathnure, 2025-02-04:
> LookupError: Setuptools.scm was unable to detect version for D:......\ nanovna-saver-main.

do you have git cli installed? Can you execute `git status` ftom the root of repository? Asking because settuptools-scm use git tools to datact current veraion

PA3CCE, 2025-02-04:
No, I have''nt, will try immedeately, thanks..

PA3CCE, 2025-02-04:
Now I have git version 2.47.1 installed, so sorry, same behaviour .

PA3CCE, 2025-02-04:
After installing git cli version 2.66.1, no change !

redrathnure, 2025-02-04:
hm.. looks like some preparation steps were missed. So, let. s start from basics. Could you post an output of following commands, including full error message:
* git version
* python --version
* uv version
* uv run python --version
* uv sync
* uv run pip list
* uv build 
* uv run NanoVNASaver

PA3CCE, 2025-02-05:
***** Git version :  
2.47.1.windows.2

***** Python version  : 
version  3.13.1

***** uv version  
uv 05.25 (9c07c3fc5 25-01-28)

***** 
uv run python -- version : 

See earlier messages : 
"Failed to generate package metadata for 'nanovnasaver==0.6.8.post1.dev19+gcdae804.d20250201 #aditable2+.'
   caused by:  the build backend returned an error
  caused by:  call to 'setuptools_wrapper.build_editable' failed (editcode1)
 [stderr]
Trackback (most recent call last):
File "string", line14, in <module>
    requires=get_requires_for_build({})
File "C:\xxx\xxx\AppData\local\uv\cache\builds-v0\.tmp 0tH94a\lib\site-packages\setuptools\build_meta.py",  in requires_for_build_executable
   return self.get_requires_for_build_wheel (config_settngs)
                                                                    ^^^^^^^^^^
            ^^^^^^^^^^^^^^^^^^^^^^

=====================================
!!!!! NOTE:  I FOUND THAT , ON MY SYSTEM  "C:\xxx\xxx\AppData\local\uv\cache\builds-v0" ENDS HERE, SO "tmp0tH94a\lib\site-packages\setuptools\build_meta.py"  DOES NOT EXIST  HERE !!!!
=====================================

***** uv sync :  
Same output as uv run python

***** uv run pip list:
Same output as uv run python

***** Pip list:
annotated-types 0.7.0
anuio 4.8.0
certifi 2024.12.14
charset-normalizer  3.4.1
cloudflare 4.0.0
configsettings 0.0.1
dependencies 7.7.1
distro 1.9.0
ez_setup 0.9
h11 0.14.0
httpcore 1.0.7
httpx 0.28.1
 idna 3.10
nunpy 2.2.2
pandas  2.2.3
pip 25.0
pydantic 2.10.6
pydantic_core 2.27.2
python-cloudflare 1.0.1
python-dateutil  2.9.0.post0
pytz 2024.2
requests 2.32.3
scikit-rf 1.5.0 
scipy 1.15.1
setuptools 75.8.0
six 1.17.0
sniffio 1.3.1
typing_extensions 4.12.2
tzdata 2025.1
urrlib3  2.3.0
uv 0.5.25
wheel 0.25.1

*****uv build:
LOT OF TEXT,  ENDING WITH  "failed to build nanovna-saver-main"

*****uv run NanoVNASaver:
LOT OF TEXT, ENDING WITH "L;ookuo error setuptools was unable to detect version for d;/yy\nanovna-saver-main"

========================================
Sorry  ccommand window does not allow copy/paste so too much text to write !!

Hope this is sufficient to find whats wrong (see my NOTE !!) 
Many thanks for your help !

redrathnure, 2025-02-05:
Okay, from unusual:
- python 13, which I have not tested (yet)
- your venv misses setuptools-scm dependency, which is necessury for the build.

I will double check it when back to home

PA3CCE, 2025-02-06:
Yes, I see that version 3.13 is missing in uv.lock.

redrathnure, 2025-02-06:
hm.. it does not support 3.13 yet. So, could you try with python 3.12?

PA3CCE, 2025-02-07:
hm.. it does not support 3.13 yet. So, could you try with python 3.12?

Sure I ccould,  but then I have to install 3.12,  and force to use it or delete 3.13.

Ain't it possible, or lot of work to adapt to 13 ?

PA3CCE, 2025-02-07:
OK, inststalled Python version 3.12, and left 3.13.1 on my system.
Now, when I issue command "python --version", I get : "Python 3.12.0".
When I ussue command "pip --version" I get  : "pip 23.2.1 from D:\Python3.12\lib\site-packages\pip (python 3.12)"
So it looks like no need to force using Python version 3.12 !

When I run "uv run nanovnasaver" , I get the same output as when using Python version 3.13.1  !!!!
Conclusion : My problem not caused by new version !

What to do now ??

In the mean time I downloaded NanoVNASaver version 7.0 and will test it.

redrathnure, 2025-02-07:
Yeah, it's bit surpise for me, but yes, looks like 3.13 is a big deal:( At least for now. In short:
- [Python 3.13.2](https://www.python.org/downloads/release/python-3132/) is out, however not all python tools are ready for it.
- meanwhile I will try to find some workaround for it, however "just downgrade" to 3.12 looks like pragmatic solution, at least for now
- BTW uv is able to install python itself. Please check `uv python list` and `uv python install` commands.
- `installed Python version 3.12, and left 3.13.1` perhaps .venv still use "the first version". Perhaps you should drop `.venv` folder and repeat the process from scratch. Please use `uv run ` prefix to check used environment (e.g. `uv run python --version` and `uv run pip --version`). And yeah, I know, this is bit messy, some tools are too smart :)
- ` I downloaded NanoVNASaver version 7.0 and will test it.` - yes, it's the best option for now. Looks like this release has all fresh (LiteVNA related) changes, so it's should be enough to prove original topic for this discussion.