NanoVNA-Saver/nanovna-saver #769: The soft crashed trying add markers

kidekoop2016, 2025-01-23:
**Describe the bug**
The soft crashed trying add markers, before and acter connect the nano na v2 Pro4 in win 10 with nanosaver 0.6.8

**To Reproduce**
Steps to reproduce the behavior:
1. Go to nano-vna. exe and open
2. Click on 'Display setup'
3. Scroll down to 'Markers' and click on Add
4. See error

**Expected behavior**
Add marker 4, 5, 6,...

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: win10
 - Python version: [if applicable]
 - NanoVNA-Saver version: 0.6.8

**Additional context**
Add any other context about the problem here.

patdoss, 2025-01-31:
Hi,
Same error found with release 0.6.9
The issue in the code is located in "DisplaySettings.py" 

The argument in new_marker.setScale is wrong:
new_marker.setScale(self.app.scaleFactor)
"self.app.scaleFactor" need to be replaced by "self.app.scale_factor"

def addMarker(self) -> None:
        new_marker = Marker("", self.app.settings)
        # wrong argument 
        # new_marker.setScale(self.app.scaleFactor)

        # New argument
        new_marker.setScale(self.app.scale_factor)

        self.app.markers.append(new_marker)
        self.app.marker_data_layout.addWidget(new_marker.get_data_layout())
        self.app.marker_frame.adjustSize()
.....

After this modification, Display setup-> Markers -> Add is functionnal (tested also Remove)
Thanks and best regards,
73's Patrick F6EXL

[image]

closed 2025-02-06 by zarath

kidekoop2016, 2025-02-07:
Problem fixed in the version 0.7.0¡
Thank you very much for your fast response.
Kind regards