NanoVNA-Saver/nanovna-saver #788: nanovna-saver.py missing from repo?

underdog, 2025-02-07:
**Describe the bug**
I have looked at the instructions many times and have seen reference to nanovna-saver.py, even on the one regarding running without install.  it appears this file is missing from the repo.  maybe it was there at the time of writing, but it doesn't exist anymore.  are there updated instructions on running without install?

**To Reproduce**
Steps to reproduce the behavior:
1. look at repo, file is missing

**Expected behavior**
Expect nanovna-saver.py file to exist in repo

**Screenshots**
N/A

**Desktop (please complete the following information):**
Repo related, not OS related

**Additional context**
None

underdog, 2025-02-07:
D	nanovna-saver.py
Date:   Tue Jan 21 18:41:23 2025 +0100

looks like it was deleted on Jan 21

zarath, 2025-02-07:
Is it still mentioned in the instructions?

uv sync && uv ui-compile && uv run NanoVNASaver

should work

underdog, 2025-02-07:
> Is it still mentioned in the instructions?
> 
> uv sync && uv ui-compile && uv run NanoVNASaver
> 
> should work

will give it a go

underdog, 2025-02-07:
> Is it still mentioned in the instructions?
> 
> uv sync && uv ui-compile && uv run NanoVNASaver
> 
> should work

so will this change the homebrew instructions also?  I'm using brew also and I found that NanoVNASaver.py  in the src directory, but it is going to fail because that's not the configured py file

zarath, 2025-02-07:
I don't use brew, but

brew install uv

and then previous instructions?

underdog, 2025-02-07:
uv ui-compile won't work.  there is a ui_compile.py in the src/ directory.  seems that the current repo has had some changes for instructions.  maybe I'll wait for instructions to be updated.

zarath, 2025-02-07:
Ok. Misstyped be me.

uv run task ui-compile

underdog, 2025-02-07:
> Ok. Misstyped be me.
> 
> uv run task ui-compile

awesome...ok, that worked now..the UI came up after running the task compile command and then the run command

redrathnure, 2025-02-07:
@zarath WDYT about just publish the python package to the PyPi repo? It would simplify answer to this and similar questions, like `pipx install nanovna-saver` then just exec `nanovna-saver`. uv is lightware and fast, however I am not sure "regular" users need to install whole build chain.

zarath, 2025-02-07:
As soon as I have time and need :-)

clytle374, 2025-04-19:
Please update the instructions, this was the answer I was looking for

jeffsf, 2025-06-06:
https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/docs/INSTALLATION.md#running-on-linux-without-installation 
is woefully incorrect. 

There is no such file to run and the similarly named file is not runnable
```
% python3 src/NanoVNASaver/NanoVNASaver.py 
Traceback (most recent call last):
  File "/Users/jeff/source/nanovna-saver/src/NanoVNASaver/NanoVNASaver.py", line 28, in <module>
    from .About import VERSION
ImportError: attempted relative import with no known parent package
```