NanoVNA-Saver/nanovna-saver #790: bug: NanoVNA-Saver version 7.0 -marker setting

PA3CCE, 2025-02-08:
# Bug Report

**NanoVNA-Saver version:** 
Version 7.0

**Current behavior:**
 Marker(s) can't be set to a desired frequency.

After starting NanoVNA-Saver and connecting it to my device "LiteVNA",  all 3 markers in the main window are set to 200MHz.
When I select a marker, set it for the desired frequency and confirm the setting by "Enter" or select another marker to  set, the marker that was set returns to the default setting of 200MHz.

Expected behaviour; 
It should be possible to set the markers individually  to a desired frequency.

PA3CCE, 2025-02-09:
OK, I see now, after sweeping you can select a marker and position it on the screen, the frequency set for is shown in the main window.
Setting the frequency manually before the sweep would be nice but not a real issue.

zarath, 2025-02-09:
I haven't found out why it doesn't accept input a start, but I was able to enter values, after changing the radiobuttons beside the marker input fields.

PA3CCE, 2025-02-09:
OK, but you can set ONE marker only, after setting it and selecting the next marker to set, the previous one returns to the default value (200MHz).

zarath, 2025-02-10:
Even this is more or less random... It seems if you change radiobuttons some more times it works prior scanning or loading of data.

PA3CCE, 2025-02-10:
Yes correct, when you set a marker and  click on it's radiiobuton several times, you can set the next one without the impact on the first one,  strange.
I have seen more proofs of instability of the software !