NanoVNA-Saver/nanovna-saver #796: bug: NanoVNA-Saver V 7.1 unstable.

PA3CCE, 2025-02-11:
# Bug Report

**NanoVNA-Saver version:** Version 7.1

**Current behavior:**
I downloaded NanoVNASaverx64.zip and unzipped the .exe file in a new repository on my Windows11 computer.
It looked like it was running well, so I tried to run a calibration 10MHz to 6GHz.

During this process, for some reason, I got the errormessage  "Error during sweep,   Stopped,   Invalid data during sweep".

Since then, when I start NanoVNA-Saver,  I can connect NanoVNA-Saver to my LiteVNA64, the device switches into USB mode,
but I can't enter the "Start" and "Stop"  frequency into NanoVNA-Saver.

Then, after a long time of waiting the earlier mentioned errormessage pops-up again.
After clicking "ÖK" I cann enter the start and stop frequency and run a sweep one time only, the second time, after waiting long, the 
errormessage shows-up again.

I tried ,  deleting NanoVNA-Saver.exe and restarting the computer,  cleaning my system by using "CCleaner PRO", re-downloading 
NanoVNASaverx64.zip and unzipping the .exe into it's repository, nothing helps, te same things happen !!

zarath, 2025-02-11:
Can you use debug mode and post output?

Reinstallation should not be necessary - states from
previous runs are in a flie called
NanoVNASaver.ini

Location of this file is mentioned in debug message

PA3CCE, 2025-02-11:
previous runs are in a flie called   NanoVNASaver.ini
==================

I had looking for it, could not find it, where isit located ??

PA3CCE, 2025-02-12:
Finally found nanovnasaver.ini in xxx/AppdData/Roaming and deleted it.
Started the software, but still get the errormessage , new  nanovnasaver.ini was generated.
Will do more testing tomorrow.

zarath, 2025-02-13:
You can call `nanovna-saver -d ` to get debug output or `nanovna-saver -D debug.txt` to create a debug file.

This file can be attached to a issue post on github, so it doesn't clutter up the issue text.

zarath, 2025-02-13:
and your .exe did not accept parameters when started from a terminal?

PA3CCE, 2025-02-13:
No, Nanovna-saver.exe on the terminal starts the program, and nanovna-saver.exe -d debug.txt just returns to the prompt and no indication that debug.txt is generated, program not started.
Sorry

zarath, 2025-02-14:
For debugging to file you need a capital -D not a small letter.

-h lists all options.

PA3CCE, 2025-02-15:
For debugging to file you need a capital -D not a small letter.

---------------------------

OK, understood.
Another questiion : is the LiteVNA's  "ENHANVED RESPONSE" option included in Nn0VNASaver ? Sofar I did not see it, or is it set to always "ON" ?    ....Is it set to on during calibration ?

PA3CCE, 2025-02-15:
I did a lot of testing today and to my surprise I found a debug file in the repository of nanovnasaver.exe it was unreportedly stored by windows CMD.  I have added it to this post.
I hope this will help to solve the problem.
If I'm correct it just shows all settings before it starts the program, it did not start it, the error occurs if it's been started at the moment when it is connected to the device, I don't know what's happening then, so I doubt it.

[debug.txt](https://github.com/user-attachments/files/18812643/debug.txt)

PA3CCE, 2025-02-16:
After having sent the rquested debug.txt file,  I think I can't do much more to help finding the cause of the problem in running Nanovna-saver together with my LiteVNA64.    When I can do more, please ask.  

Please answer my cuestion about using the "calibration".  Are all settings fixed, "Ënhanced RESPONSE" included, or is it possible to include some user settings like "nummber of segments" , number of "ÄVG's" etc.

Thank you so much  for all the hard work to improve the software.

zarath, 2025-02-17:
Hi Jos, 

unluckily there is no crash to be seen in the attached debug.txt.

If you have trouble with restarting you can delete:

C:/Users/Jos/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini

and any other .ini file in that directory to get a fresh / like new installed state.

Can you give the program another try with the -D newdebug.txt option started from the console?

PA3CCE, 2025-02-17:
unluckily there is no crash to be seen in the attached debug.txt.
===============
Yes that's what I said.

C:/Users/Jos/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
==================
Deleted it several times.

Can you give the program another try with the -D newdebug.txt option started from the console?
===============================
That's what I'm doing at the moment, will send it when I find an interesting one !

PA3CCE, 2025-02-17:
OK, I started  " nanovnasaver.exe -D debug-4.txt" in Window's CMD (had NOT deleted .ini )
NanoVNASavr  started up.
Then I clicked "Connect".
I had to wait abt 75 seconds before the wellknown errormessage came up.
Then I could enter Start 10M and Stop 100M.
I left all settings in default.
Cliccked "Calibrati.on, Calibration assistant"
Started the "Short" calibration.
Had to wait long before errormewssage "Invalid data" came up.

This one showes more :
[debug-4.txt](https://github.com/user-attachments/files/18833177/debug-4.txt)

PA3CCE, 2025-02-18:
Today I did the same thing, but after deleting nanovnasaver.ini first, the short calibration ended up with
 "List assignment index out of range " , program hangs.

[debug-6.txt](https://github.com/user-attachments/files/18845959/debug-6.txt)

Then deleted nanovnasaver.ini and left Start/Stop 200MHz/400MHz.
Calibration "Load" ended up with "Invalid data during sweep" and hangs

[debug-8.txt](https://github.com/user-attachments/files/18846013/debug-8.txt)

zarath, 2025-02-18:
```
2025-02-18 12:57:56,662 - NanoVNASaver.Hardware.NanoVNA_V2 - INFO - reading values
2025-02-18 12:58:03,853 - NanoVNASaver.Hardware.NanoVNA_V2 - WARNING - expected 6432 bytes, got 5825
2025-02-18 12:58:08,245 - NanoVNASaver.Hardware.NanoVNA_V2 - DEBUG - Freq index from: 48
2025-02-18 12:58:08,245 - NanoVNASaver.SweepWorker - ERROR - list assignment index out of range
```
```
2025-02-18 14:06:00,767 - NanoVNASaver.Hardware.NanoVNA_V2 - WARNING - expected 6432 bytes, got 928
```

The problem is in the serial communication with the LiteVNA64 (it's derived from NanaVNA_V2 protocol therefor the modules log name).  It seems that for some reason not all bytes are recieved on all commands...

PA3CCE, 2025-02-18:
The communication was working when the original V2 protocol was used (when LiteVNA64 was recognised as being a V2)!

PA3CCE, 2025-02-19:
Did the same thing using the new version 7.2.
After starting had to wait long before the errormessage cam up again.
Then I could finisch a complete SOLT calibration 10-100MHz.
After that I could run a trough sweep  6 times,  before I got the errormessage again.
However the complete full trace was NOT at 0dB as it should.
Results attached here.

[image]

[Debug-7.2.txt](https://github.com/user-attachments/files/18871926/Debug-7.2.txt