NanoVNA-Saver/nanovna-saver #805: bug: bug: 0.7.2 Can't select a band in the 'Sweep setting' dialogue.

barjac, 2025-02-22:
# Bug Report
The band selector does not list any band to select, it only changes to blue background.

**NanoVNA-Saver version:**
commit cc7c91dd714c9bb728929f1fd506e70cc54c802f (HEAD, tag: v0.7.2)

**Current behavior:**

As title.**Expected behavior:**
The combo box should list to bands that are in the default configuration file.

**Steps to reproduce:**
Start the program with or without connecting to a VNA and try to set the band sweep.

**Related code:**
n/a

**Other information:**
I reported this for 0.7.1 but the 'fix' was pushed and the bug report closed at the same time before I had any chance to test.
Please try again but give me time to test and report back before closing :)

Thanks.

barjac, 2025-02-23:
I have discovered during more testing that this does not happen when using an XFCE session but it does using KDE Plasma 6.

barjac, 2025-02-24:
There are no error messages in a terminal.

Here is the situation on launch with a fresh .ini (where does the 135.7kHz and 137.8kHz come from?):

[image]

 On left clicking the band select combo box:

[image]

If the Set band sweep is clicked then the 135.7kHz and 137.8kHz appear as the start/stop frequencies. The 50kHz and 35MHz are from the connected nanovna.

zarath, 2025-02-24:
Can you check my newest branch / PR

https://github.com/zarath/nanovna-saver/tree/fix/sweep_band_selection

barjac, 2025-02-24:
> Can you check my newest branch / PR
> 
> https://github.com/zarath/nanovna-saver/tree/fix/sweep_band_selection

Perfect! Tested in 3 D.E.s and working fine in all including Plasma6 :)

Many thanks!

Is a new version with this fix imminent or should I patch our 0.7.2?

zarath, 2025-02-24:
I publish a new release immediately

barjac, 2025-02-25:
> I publish a new release immediately

Super,
Thanks again!

closed 2025-02-27 by zarath