NanoVNA-Saver/nanovna-saver #808: bug: 0.7.3 Can't install with make deb : setup.py No such file or directory

moricef, 2025-02-27:
# Bug Report

**NanoVNA-Saver version:**

 0.7.3

**Current behavior:**

 $ make deb
python3 setup.py clean
python3: can't open file '/nanovna-saver-0.7.3/setup.py': [Errno 2] No such file or directory
make: *** [Makefile:40: clean] Error 2

**Expected behavior:**

**Steps to reproduce:**

**Related code:**

```
insert short code snippets here
```

**Other information:**

AdamC19, 2025-03-06:
Just noting that I'm encountering something similar, unfortunately.

I first clone the repo master (currently at v0.7.3), then:
```
$ make deb
python3 setup.py clean
python3: can't open file '/home/AdamC19/Documents/nanovna-saver/setup.py': [Errno 2] No such file or directory
make: *** [Makefile:40: clean] Error 2
```

Maybe worth noting that I'm running this all in a python virtual env, however I'm quite confident that I have all the dependencies installed, since I've tried several of the documented methods to install (or even just run) and so far nothing has worked. 

I might be missing something, but it seems like there does not exist a `setup.py` file in the expected place?

I'm running Raspberry Pi OS on a Raspvberry pi 5.

mtnredhed, 2025-03-18:
In my case, I'm on Linux Mint 22.1 and python 3.12.3

Abdull, 2025-03-19:
I have tried every single described installation instruction presented on [`docs/INSTALLATION.md`](https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/docs/INSTALLATION.md), and none worked for my system (Debian 12, Framework 13 AMD laptop).

<details>
<summary>almost all unsuccessful attempts (and the final successful attempt 5) (I spare you three more unsuccessful attempts) (click to open)</summary>
## 1

```
# taken verbatim from INSTALLATION.md

$ pip3 install https://github.com/NanoVNA-Saver/nanovna-saver/archive/refs/tags/v0.5.5.tar.gz
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or
OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

$ python3 -m venv venv
$ source venv/bin/activate
(venv) $ pip3 install https://github.com/NanoVNA-Saver/nanovna-saver/archive/refs/tags/v0.5.5.tar.gz

(venv) $ NanoVNASaver
# works, but is old version, v0.5.5
```

## 2

```
# adapting "pip3 install .tar.gz ('tgz'?)" method for latest version.
# head over to https://github.com/NanoVNA-Saver/nanovna-saver/releases
# latest release (as of this writing) https://github.com/NanoVNA-Saver/nanovna-saver/releases/tag/v0.7.3 does not have a .tar.gz release
# older releases also don't list .tar.gz files; only .zip files.
# cleverly adapting original example command URL for latest release:

$ python3 -m venv venv
$ source venv/bin/activate
(venv) $ pip3 install https://github.com/NanoVNA-Saver/nanovna-saver/archive/refs/tags/v0.7.3.tar.gz

Collecting https://github.com/NanoVNA-Saver/nanovna-saver/archive/refs/tags/v0.7.3.tar.gz
  Using cached https://github.com/NanoVNA-Saver/nanovna-saver/archive/refs/tags/v0.7.3.tar.gz
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [64 lines of output]
      Traceback (most recent call last):
# ...
      LookupError: setuptools-scm was unable to detect version for /tmp/pip-req-build-ljp10wwc.

      Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't
 contain the necessary metadata and will not work.

      For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj
[... 187 lines cut ...]
## TL;DR

**Forget about [`docs/INSTALLATION.md`](https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/docs/INSTALLATION.md). Trying anything from it will cost you half a working day. That file should be deleted until fixed.**

These are the steps of getting v0.7.3 to build and run under Debian 12 bookworm:

```
# prerequisites
sudo apt install pipx
pipx install uv

# clone, build, and run
git clone https://github.com/NanoVNA-Saver/nanovna-saver
cd nanovna-saver
uv venv
source .venv/bin/activate
uv sync
uv build

# when inside venv, to start NanoVNASaver, you can use the following command:
NanoVNASaver

# when not inside venv, to start NanoVNASaver, you have to use the following command:
uv run NanoVNASaver
```

u4y0u, 2025-07-31:
@Abdull 

It works if you run the command with sudo
 sudo NanoVNASaver

Otherwise

NanoVNASaver/NanoVNASaver.py", line 123, in __init__
    self.serial_control = SerialControl(self)
NanoVNASaver/Controls/SerialControl.py", line 46, in __init__
    self.rescanSerialPort()
ontrols/SerialControl.py", line 83, in rescanSerialPort
    for iface in get_interfaces():
NanoVNASaver/Hardware/Hardware.py", line 120, in get_interfaces
    iface.open()
.venv/lib/python3.12/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyACM0: [Errno 13] Permission non accordée: '/dev/ttyACM0'