NanoVNA-Saver/nanovna-saver #811: bug: [Errno 2] No such file - Linux Mint 22.1 Xia [installation]

n1moy, 2025-03-05:
# Bug Report

**NanoVNA-Saver version:**

**Current behavior:**

**Expected behavior:**

**Steps to reproduce:**

**Related code:**

```
insert short code snippets here
```

**Other information:**

wolframalexa, 2025-05-07:
I am having this issue on Ubuntu 22.04 as well. I am following the installation instructions here: https://github.com/NanoVNA-Saver/nanovna-saver/blob/main/docs/INSTALLATION.md#ubuntu-2004--2204--2404

My output:
```
(.venv_nano) alexa@parrot:~/rf/nanovna-saver  (main)$  sudo apt install python3 python3-pip libxcb-cursor-dev
 python3 -m venv ~/.venv_nano
 . ~/.venv_nano/bin/activate
 pip install -U pip
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libxcb-cursor-dev is already the newest version (0.1.1-4ubuntu1).
python3 is already the newest version (3.10.6-1~22.04.1).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.5).
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
Requirement already satisfied: pip in /home/alexa/.venv_nano/lib/python3.10/site-packages (25.1.1)
(.venv_nano) alexa@parrot:~/rf/nanovna-saver  (main)$  python3 -m pip install .
Processing /home/alexa/rf/nanovna-saver
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: pyserial~=3.5 in /home/alexa/.venv_nano/lib/python3.10/site-packages (from NanoVNASaver==0.7.3.post1.dev5+g25e84b4.d20250507) (3.5)
Requirement already satisfied: pyside6~=6.8 in /home/alexa/.venv_nano/lib/python3.10/site-packages (from NanoVNASaver==0.7.3.post1.dev5+g25e84b4.d20250507) (6.9.0)
Requirement already satisfied: numpy~=2.1 in /home/alexa/.venv_nano/lib/python3.10/site-packages (from NanoVNASaver==0.7.3.post1.dev5+g25e84b4.d20250507) (2.2.5)
Requirement already satisfied: scipy~=1.14 in /home/alexa/.venv_nano/lib/python3.10/site-packages (from NanoVNASaver==0.7.3.post1.dev5+g25e84b4.d20250507) (1.15.2)
Requirement already satisfied: shiboken6==6.9.0 in /home/alexa/.venv_nano/lib/python3.10/site-packages (from pyside6~=6.8->NanoVNASaver==0.7.3.post1.dev5+g25e84b4.d20250507) (6.9.0)
Requirement already satisfied: PySide6-Essentials==6.9.0 in /home/alexa/.venv_nano/lib/python3.10/site-packages (from pyside6~=6.8->NanoVNASaver==0.7.3.post1.dev5+g25e84b4.d20250507) (6.9.0)
Requirement already satisfied: PySide6-Addons==6.9.0 in /home/alexa/.venv_nano/lib/python3.10/site-packages (from pyside6~=6.8->NanoVNASaver==0.7.3.post1.dev5+g25e84b4.d20250507) (6.9.0)
Building wheels for collected packages: NanoVNASaver
  Building wheel for NanoVNASaver (pyproject.toml) ... done
  Created wheel for NanoVNASaver: filename=nanovnasaver-0.7.3.post1.dev5+g25e84b4.d20250507-py3-none-any.whl size=197498 sha256=373fb257096d00ec37250ef6e3a7a9c7f531db2a2b79ac111f4e9a14d672d106
  Stored in directory: /home/alexa/.cache/pip/wheels/b8/63/34/5bb2d08f74167e3fc80208a1d674f668ea870176e0290326c6
Successfully built NanoVNASaver
Installing collected packages: NanoVNASaver
  Attempting uninstall: NanoVNASaver
    Found existing installation: NanoVNASaver 0.7.3.post1.dev5+g25e84b4.d20250507
    Uninstalling NanoVNASaver-0.7.3.post1.dev5+g25e84b4.d20250507:
      Successfully uninstalled NanoVNASaver-0.7.3.post1.dev5+g25e84b4.d20250507
Successfully installed NanoVNASaver-0.7.3.post1.dev5+g25e84b4.d20250507
(.venv_nano) alexa@parrot:~/rf/nanovna-saver  (main)$  . ~/.venv_nano/bin/activate
 python3 nanovna-saver.py
python3: can't open file '/home/alexa/rf/nanovna-saver/nanovna-saver.py': [Errno 2] No such file or directory
(.venv_nano) alexa@parrot:~/rf/nanovna-saver  (main)$ NanoVNASaver
NanoVNASaver      NanoVNASaver-gui  
(.venv_nano) alexa@parrot:~/rf/nanovna-saver  (main)$ NanoVNASaver
NanoVNASaver      NanoVNASaver-gui  
(.venv_nano) alexa@parrot:~/rf/nanovna-saver  (main)$ NanoVNASaver-gui 
Traceback (most recent call last):
  File "/home/alexa/.venv_nano/bin/NanoVNASaver-gui", line 5, in <module>
    from NanoVNASaver.__main__ import main
  File "/home/alexa/.venv_nano/lib/python3.10/site-packages/NanoVNASaver/__init__.py", line 1, in <module>
    from .NanoVNASaver import NanoVNASaver
  File "/home/alexa/.venv_nano/lib/python3.10/site-packages/NanoVNASaver/NanoVNASaver.py", line 68, in <module>
    from .Windows import (
  File "/home/alexa/.venv_nano/lib/python3.10/site-packages/NanoVNASaver/Windows/__init__.py", line 1, in <module>
    from .About import AboutWindow
  File "/home/alexa/.venv_nano/lib/python3.10/site-packages/NanoVNASaver/Windows/About.py", line 31, in <module>
    from .ui.about import Ui_DialogAbout
  File "/home/alexa/.venv_nano/lib/python3.10/site-packages/NanoVNASaver/Windows/ui/__init__.py", line 4, in <module>
    from .about import Ui_DialogAbout
ModuleNotFoundError: No module named 'NanoVNASaver.Windows.ui.about'
(.venv_nano) alexa@parrot:~/rf/nanovna-saver  (main)$ NanoVNASaver
NanoVNASaver      NanoVNASaver-gui  
(.venv_nano) alexa@parrot:~/rf/nanovna-saver  (main)$ NanoVNASaver
NanoVNASaver      NanoVNASaver-gui  
(.venv_nano) alexa@parrot:~/rf/nanovna-saver  (main)$ NanoVNASaver
Traceback (most recent call last):
  File "/home/alexa/.venv_nano/bin/NanoVNASaver", line 5, in <module>
    from NanoVNASaver.__main__ import main
  File "/home/alexa/.venv_nano/lib/python3.10/site-packages/NanoVNASaver/__init__.py", line 1, in <module>
    from .NanoVNASaver import NanoVNASaver
  File "/home/alexa/.venv_nano/lib/python3.10/site-packages/NanoVNASaver/NanoVNASaver.py", line 68, in <module>
    from .Windows import (
  File "/home/alexa/.venv_nano/lib/python3.10/site-packages/NanoVNASaver/Windows/__init__.py", line 1, in <module>
    from .About import AboutWindow
  File "/home/alexa/.venv_nano/lib/python3.10/site-packages/NanoVNASaver/Windows/About.py", line 31, in <module>
    from .ui.about import Ui_DialogAbout
  File "/home/alexa/.venv_nano/lib/python3.10/site-packages/NanoVNASaver/Windows/ui/__init__.py", line 4, in <module>
    from .about import Ui_DialogAbout
ModuleNotFoundError: No module named 'NanoVNASaver.Windows.ui.about'
```

jeffsf, 2025-06-06:
Same ModuleNotFoundError with pip install into a venv on macOS

```
(nanovna-saver) jeff@mini ...(1)/jeff/source/nanovna-saver % ~/venv/nanovna-saver/bin/NanoVNASaver-gui 
Traceback (most recent call last):
  File "/Users/jeff/venv/nanovna-saver/bin/NanoVNASaver-gui", line 5, in <module>
    from NanoVNASaver.__main__ import main
  File "/Users/jeff/venv/nanovna-saver/lib/python3.13/site-packages/NanoVNASaver/__init__.py", line 1, in <module>
    from .NanoVNASaver import NanoVNASaver
  File "/Users/jeff/venv/nanovna-saver/lib/python3.13/site-packages/NanoVNASaver/NanoVNASaver.py", line 68, in <module>
    from .Windows import (
    ...<8 lines>...
    )
  File "/Users/jeff/venv/nanovna-saver/lib/python3.13/site-packages/NanoVNASaver/Windows/__init__.py", line 1, in <module>
    from .About import AboutWindow
  File "/Users/jeff/venv/nanovna-saver/lib/python3.13/site-packages/NanoVNASaver/Windows/About.py", line 31, in <module>
    from .ui.about import Ui_DialogAbout
  File "/Users/jeff/venv/nanovna-saver/lib/python3.13/site-packages/NanoVNASaver/Windows/ui/__init__.py", line 4, in <module>
    from .about import Ui_DialogAbout
ModuleNotFoundError: No module named 'NanoVNASaver.Windows.ui.about'
(nanovna-saver) jeff@mini ...(1)/jeff/source/nanovna-saver % ~/venv/nanovna-saver/bin/NanoVNASaver    
Traceback (most recent call last):
  File "/Users/jeff/venv/nanovna-saver/bin/NanoVNASaver", line 5, in <module>
    from NanoVNASaver.__main__ import main
  File "/Users/jeff/venv/nanovna-saver/lib/python3.13/site-packages/NanoVNASaver/__init__.py", line 1, in <module>
    from .NanoVNASaver import NanoVNASaver
  File "/Users/jeff/venv/nanovna-saver/lib/python3.13/site-packages/NanoVNASaver/NanoVNASaver.py", line 68, in <module>
    from .Windows import (
    ...<8 lines>...
    )
  File "/Users/jeff/venv/nanovna-saver/lib/python3.13/site-packages/NanoVNASaver/Windows/__init__.py", line 1, in <module>
    from .About import AboutWindow
  File "/Users/jeff/venv/nanovna-saver/lib/python3.13/site-packages/NanoVNASaver/Windows/About.py", line 31, in <module>
    from .ui.about import Ui_DialogAbout
  File "/Users/jeff/venv/nanovna-saver/lib/python3.13/site-packages/NanoVNASaver/Windows/ui/__init__.py", line 4, in <module>
    from .about import Ui_DialogAbout
ModuleNotFoundError: No module named 'NanoVNASaver.Windows.ui.about'
```

thehans, 2025-07-18:
I got it to build and run following the instructions at the end "TL;DR" section of this comment 
https://github.com/NanoVNA-Saver/nanovna-saver/issues/808#issuecomment-2738034196