NanoVNA-Saver/nanovna-saver #812: bug: ModuleNotFoundError: No module named 'NanoVNASaver.Windows.ui.about'

hvegh, 2025-03-08:
**NanoVNA-Saver version:**

Current git

Switched to *uv*:  

uv run NanoVNASaver
Traceback (most recent call last):
  File "/home/henk/prj/device/tools/liteVNA/nanovna-saver/.venv/bin/NanoVNASaver", line 4, in <module>
    from NanoVNASaver.__main__ import main
  File "/home/henk/prj/device/tools/liteVNA/nanovna-saver/src/NanoVNASaver/__init__.py", line 1, in <module>
    from .NanoVNASaver import NanoVNASaver
  File "/home/henk/prj/device/tools/liteVNA/nanovna-saver/src/NanoVNASaver/NanoVNASaver.py", line 68, in <module>
    from .Windows import (
  File "/home/henk/prj/device/tools/liteVNA/nanovna-saver/src/NanoVNASaver/Windows/__init__.py", line 1, in <module>
    from .About import AboutWindow
  File "/home/henk/prj/device/tools/liteVNA/nanovna-saver/src/NanoVNASaver/Windows/About.py", line 31, in <module>
    from .ui.about import Ui_DialogAbout
  File "/home/henk/prj/device/tools/liteVNA/nanovna-saver/src/NanoVNASaver/Windows/ui/__init__.py", line 4, in <module>
    from .about import Ui_DialogAbout
ModuleNotFoundError: No module named 'NanoVNASaver.Windows.ui.about'

hvegh, 2025-03-08:
Ok, I should have done:

uv sync && uv run task ui-compile && uv run NanoVNASaver

Not familiar with uv

closed 2025-03-08 by hvegh

Nannk, 2025-04-15:
Same problem here after updating from aur.