NanoVNA-Saver/nanovna-saver #831: bug: No Linux release for v0.7.3 ?

thehans, 2025-07-17:
# Bug Report

**NanoVNA-Saver version:**
v0.7.3

**Current behavior:**
https://github.com/NanoVNA-Saver/nanovna-saver/releases/tag/v0.7.3
Has no linux release

**Expected behavior:**
linux release exists

kellimac, 2025-08-08:
I was able to get this release to work for about five minutes until something else python-related broke it. None of the installation and run instructions work at all on Linux Mint. 

NanoVNASaver used to install and work pretty well. Now it seems useless. I'm not new to any of this. This release is just broken broken broken.

flocentblack, 2025-08-10:
Thanks kellimac for pointing me to @Abdull's instuctions which got it working for me.